uzh-rpg/rpg_esim #6: library called `ze` keeps giving error.

RunqiuBao, 2018-11-12:
Hi, what is this library called `ze`? Does it have similar functions as eigen?
It keeps giving me error.
[image]

supitalp, 2018-11-13:
Hi, `ze` is a set of useful functions provided by https://github.com/uzh-rpg/ze_oss
What error do you have exactly? On which system?

supitalp, 2018-12-06:
Hello,
Did you resolve this problem? If not, could you please give me a bit more information about it (which error do you have exactly, which system, version of ROS, CMake and gcc?
Thanks.

supitalp, 2019-03-22:
Hi,
Since you did not reply, I assume the problem is fixed and will close this issue.

closed 2019-03-22 by supitalp