uzh-rpg/rpg_esim #7: Suggestion to change in the installation

duguyue100, 2018-11-16:
Hi, I just compiled the ESIM on my Ubuntu 16.04 machine with ROS Kinetic. I just recently setup my machine so the OS is rather clean. I noticed two problems during the installation:

First, In this line: https://github.com/uzh-rpg/ze_oss/blob/master/ze_cmake/cmake/modules/ze_setup.cmake#L4

The option `Werror` is turned on so that some warnings in the `pyconfig.h` will be flagged as errors. With this, I couldn't compile `ze_matplotlib`

Second, while I was compiling `ze_splines`, it gives me the following errors:

```
/usr/lib/x86_64-linux-gnu/libapr-1.so.0: undefined reference to `uuid_generate@UUID_1.0'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dgywork/catkin_ws/devel/lib/ze_splines/viz_splines_test] Error 1
make[1]: *** [CMakeFiles/viz_splines_test.dir/all] Error 2
make: *** [all] Error 2
```

Turns out the version of `libuuid` supported by the Ubuntu 16.04 is too old. Since I have a conda on my machine, so I installed `libuuid` through conda to fix it:

```
conda install -c conda-forge libuuid 
```

Yuhuang.

supitalp, 2018-12-06:
Hello,
Thanks for your feedback. I will leave the issue open for now in case anyone else runs into the same problem. I have never seen this problem, although I tested ESIM on the same system (Ubuntu 16.04 + ROS Kinetic). For reference, could you please tell me what version of Python, CMake, and gcc you are using? 
Thanks.

charleschen6, 2019-06-18:
I also met with same error by execute catkin_make (http://wiki.ros.org/catkin/Tutorials/workspace_overlaying 2.3Building Your catkin Workspace)
Python version: 2.7.16
CMake: 3.5.1
GCC: 6.5.0
 libuuid-perl                                 0.24-1build1
libuuid1:amd64                               2.27.1-6ubuntu3.7

duguyue100, 2019-06-18:
Hi, @charleschen6 are you using the Python supplied by Ubuntu, Virtual environment, or Conda?
In my case, because I link ROS with Conda's Python, so I can use libuuid in Conda.
Otherwise, maybe build the newer library from scratch?

Regards,
Yuhuang.

guanmeng123, 2019-11-22:
Python version: 2.7.12
CMake: 3.5.1
GCC: 5.4.0
Ubuntu 16.04 + ROS Kinetic

I also met with same error by execute catkin_make

/usr/lib/x86_64-linux-gnu/libapr-1.so.0: undefined reference to `uuid_generate@UUID_1.0'
collect2: error: ld returned 1 exit status
interactive_marker_twist_server/CMakeFiles/interactive_marker_twist_server.dir/build.make:119: recipe for target '/home/student908/wuda/guanmeng/DRL_Path_Planning/devel/lib/interactive_marker_twist_server/marker_server' failed
make[2]: *** [/home/student908/wuda/guanmeng/DRL_Path_Planning/devel/lib/interactive_marker_twist_server/marker_server] Error 1
CMakeFiles/Makefile2:5350: recipe for target 'interactive_marker_twist_server/CMakeFiles/interactive_marker_twist_server.dir/all' failed
make[1]: *** [interactive_marker_twist_server/CMakeFiles/interactive_marker_twist_server.dir/all] Error 2

I'm looking forward to your help,thanks

ErnestDeiven, 2019-12-06:
> Python version: 2.7.12
> CMake: 3.5.1
> GCC: 5.4.0
> Ubuntu 16.04 + ROS Kinetic
>
> I also met with same error by execute catkin_make
>
> /usr/lib/x86_64-linux-gnu/libapr-1.so.0: undefined reference to [...]

I have soleved the problem by downloading new flycapture2 to reinstall https://github.com/ErnestDeiven/flycapture2

smmmmi, 2020-09-07:
Possible solution fro libuuid:

`It is good to remove the libuuid library file under anaconda2/lib, and finally solved this problem.
ls ~/anaconda2/lib/libuuid*
mkdir ~/anaconda2/libuuid
mv ~/anaconda2/lib/libuuid* ~/anaconda2/libuuid
`
From [here](https://www.programmersought.com/article/97791874444/)

ngkhiem97, 2020-10-11:
> Possible solution for libuuid:
> 
> "It is good to remove the libuuid library file under anaconda2/lib, and finally solved this problem."
> `ls ~/anaconda2/lib/libuuid*`
> `mkdir ~/anaconda2/libuuid`
> `mv ~/anaconda2/lib/libuuid* ~/anaconda2/libuuid`
> From [here](https://www.programmersought.com/article/97791874444/)

You saved my day bro!

s0um0r0y, 2025-05-12:
I have a suggestion for people using the latest version of anaconda3 to use

```
ls ~/anaconda3/lib/libuuid*
mkdir ~/anaconda3/libuuid
mv ~/anaconda3/lib/libuuid* ~/anaconda3/libuuid
```

> Possible solution for libuuid:
>
> "It is good to remove the libuuid library file under anaconda2/lib, and finally solved this problem." `ls ~/anaconda2/lib/libuuid*` `mkdir ~/anaconda2/libuuid` `mv [...]