uzh-rpg/rpg_esim #9: Failed to generate to OpenGL window

Helloworld-Le, 2018-12-05:
Hi there, I am trying this new simulator on a virtual machine. After installation, I always get this error. Can you see what is the reason? Thanks in advance! (I tried with other glfw3 examples and I can successfully generate windows.)

the error shows as:**_F1205 00:37:45.703743  3456 opengl_renderer.cpp:58] Failed to create GLFW window__**

as whole:
... logging to /home/parallels/.ros/log/fb11ed2a-f825-11e8-97e8-001c42bb2f7d/roslaunch-parallels-vm-3416.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://parallels-vm:32995/

SUMMARY
========

PARAMETERS
 * /hector_trajectory_server/source_frame_name: cam0
 * /hector_trajectory_server/target_frame_name: map
 * /hector_trajectory_server/trajectory_publish_rate: 15
 * /hector_trajectory_server/trajectory_update_rate: 15
 * /optic_flow_viz/arrows_scale: 0.07
 * /optic_flow_viz/arrows_step: 7
 * /optic_flow_viz/arrows_upsample_factor: 1
 * /optic_flow_viz/publish_rate: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    hector_trajectory_server (hector_trajectory_server/hector_trajectory_server)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

auto-starting new master
process[master]: started with pid [3429]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb11ed2a-f825-11e8-97e8-001c42bb2f7d
process[rosout-1]: started with pid [3445]
started core service [/rosout]
process[esim_node-2]: started with pid [3456]
process[dvs_renderer-3]: started with pid [3467]
process[optic_flow_viz-4]: started with pid [3468]
process[hector_trajectory_server-5]: started with pid [3469]
[ INFO] [1543970265.542087462]: Waiting for tf transform data between frames map and cam0 to become available
I1205 00:37:45.645692  3456 data_provider_online_render.cpp:63] Horizontal FOV: 61.7164 deg
I1205 00:37:45.646080  3456 data_provider_online_render.cpp:64] Vertical FOV: 48.2168 deg
I1205 00:37:45.646138  3456 data_provider_online_render.cpp:65] Diagonal FOV: 73.4823 deg
### **_F1205 00:37:45.703743  3456 opengl_renderer.cpp:58] Failed to create GLFW window__**
*** Check failure stack trace: ***
    @     0x7f7b46e2943d  google::LogMessage::Fail()
    @     0x7f7b46e2b253  google::LogMessage::SendToLog()
    @     0x7f7b46e28fcb  google::LogMessage::Flush()
    @     0x7f7b46e2bc3e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f7b42e6b64a  event_camera_simulator::OpenGLRenderer::init()
    @     0x7f7b42e6bd9a  event_camera_simulator::OpenGLRenderer::setCamera()
    @     0x7f7b484087c6  event_camera_simulator::DataProviderOnlineMoving3DCameraRig::DataProviderOnlineMoving3DCameraRig()
    @     0x7f7b48400ad7  event_camera_simulator::loadDataProviderFromGflags()
    @           0x404e9e  main
    @     0x7f7b462b9830  __libc_start_main
    @           0x405819  _start
**_[esim_node-2] process has died [pid 3456, exit code -6, cmd /home/parallels/sim_ws/devel/lib/esim_ros/esim_node --v=1 --vmodule=data_provider_from_folder=10 --flagfile=/home/parallels/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/opengl.conf __name:=esim_node __log:=/home/parallels/.ros/log/fb11ed2a-f825-11e8-97e8-001c42bb2f7d/esim_node-2.log]._**
log file: /home/parallels/.ros/log/fb11ed2a-f825-11e8-97e8-001c42bb2f7d/esim_node-2*.log
[INFO] [1543970266.207507]: Started flow converter node
[INFO] [1543970266.208061]: Step size between arrows: 7
[INFO] [1543970266.208381]: Scale factor: 0.07
[INFO] [1543970266.208666]: Upsample factor: x1
[INFO] [1543970266.209042]: Publish rate: 100 Hz
[ WARN] [1543970285.548950092]: No transform between frames map and cam0 available after 20.006891 seconds of waiting. This warning only prints once.

supitalp, 2018-12-06:
Hello,
Thanks for reporting the issue. I don't know what could cause this error. Could you please try to get a more complete description of the error as described here: https://www.glfw.org/docs/latest/intro_guide.html#error_handling ?

In short, you can add this line: `glfwSetErrorCallback(error_callback);` just after line 47 for example, and add the following error callback function in the same file:

```
void error_callback(int error, const char* description)
{
  LOG(FATAL) << description;
}
```

Then, can you report the error here?

Thanks.

Helloworld-Le, 2018-12-07:
Hi, thanks for your reply. I just change to use a desktop and update to Ubuntu 18.04.1. Then there is no more errors!

closed 2018-12-07 by Helloworld-Le