uzh-rpg/rpg_esim #10: Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist

gaImdea, 2018-12-12:
I am trying to run the Photorealistic Rendering Engine that can be find at https://github.com/uzh-rpg/rpg_esim/wiki/Photorealistic-Rendering-Engine-based-on-Unreal-Engine 
I have followed all steps explained , when I  launch 

> roslaunch` esim_ros esim.launch config:=cfg/unrealcv.conf

I get the following error  on  ze_oss/ze_cameras/src/camera_rig.cpp line 106 

> F1212 12:19:18.864552 14026 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.

here is the full output

`NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    hector_trajectory_server (hector_trajectory_server/hector_trajectory_server)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

auto-starting new master
process[master]: started with pid [14000]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c381b0c2-fdff-11e8-a56e-34e6d7284e42
process[rosout-1]: started with pid [14013]
started core service [/rosout]
process[esim_node-2]: started with pid [14026]
process[dvs_renderer-3]: started with pid [14031]
process[optic_flow_viz-4]: started with pid [14032]
process[hector_trajectory_server-5]: started with pid [14033]
[ INFO] [1544613558.281283420]: Waiting for tf transform data between frames map and cam0 to become available
I1212 12:19:18.846436 14026 imu_factory.cpp:20] Initialize bias ...
I1212 12:19:18.864478 14026 imu_factory.cpp:27] done.
I1212 12:19:18.864506 14026 imu_factory.cpp:35] Initialize IMU ...
I1212 12:19:18.864513 14026 imu_factory.cpp:44] done.

F1212 12:19:18.864552 14026 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.
*** Check failure stack trace: ***
    @     0x7f9bb2d9d43d  google::LogMessage::Fail()
    @     0x7f9bb2d9f253  google::LogMessage::SendToLog()
    @     0x7f9bb2d9cfcb  google::LogMessage::Flush()
    @     0x7f9bb2d9fc3e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f9bae56cf57  ze::cameraRigFromGflags()
    @     0x7f9bb437c16b  event_camera_simulator::DataProviderOnlineMoving3DCameraRig::DataProviderOnlineMoving3DCameraRig()
    @     0x7f9bb4374a97  event_camera_simulator::loadDataProviderFromGflags()
    @           0x404e4e  main
    @     0x7f9bb222d830  __libc_start_main
    @           0x4057c9  _start
[INFO] [1544613559.038977]: Started flow converter node
[INFO] [1544613559.039281]: Step size between arrows: 7
[INFO] [1544613559.039456]: Scale factor: 0.07
[INFO] [1544613559.039638]: Upsample factor: x1
[INFO] [1544613559.039797]: Publish rate: 100 Hz

[esim_node-2] process has died [pid 14026, exit code -6, cmd /home/oaiepc/sim_ws/devel/lib/esim_ros/esim_node --v=1 --vmodule=data_provider_from_folder=10 --flagfile=/home/oaiepc/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/unrealcv.conf __name:=esim_node __log:=/home/oaiepc/.ros/log/c381b0c2-fdff-11e8-a56e-34e6d7284e42/esim_node-2.log].
log file: /home/oaiepc/.ros/log/c381b0c2-fdff-11e8-a56e-34e6d7284e42/esim_node-2*.log
[ WARN] [1544613578.283323455]: No transform between frames map and cam0 available after 20.002056 seconds of waiting. This warning only prints once.`

Any help would be really appreciated
thanks

supitalp, 2018-12-12:
Hello,
Did you make sure that the path to the calibration file is correctly set in the configuration file? (here: https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/cfg/unrealcv.conf#L15)

gaImdea, 2018-12-12:
Actually , I didn't , I have adjusted the file ,
I will also  try to interact with the scene but at the moment everything seems perfectly working.
thanks a lot as  usual

supitalp, 2018-12-12:
Great! Closing this issue then.

closed 2018-12-12 by supitalp

SeaEastXu, 2020-05-30:
$ rosrun esim_ros esim_node \ --data_source=2 \ --path_to_output_bag=~/sim_ws/tmp/out.bag \ --path_to_data_folder=/home/seaeast/sim_ws/tmp/demo/frames/ \ --ros_publisher_frame_rate=10 \ --exposure_time_ms=10.0 \ --use_log_image=1  --log_eps=0.1  --contrast_threshold_pos=0.15  --contrast_threshold_neg=0.15
F0530 16:07:07.848632  8489 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.
*** Check failure stack trace: ***
    @     0x7fe5b4757b0d  google::LogMessage::Fail()
    @     0x7fe5b47599b1  google::LogMessage::SendToLog()
    @     0x7fe5b475763d  google::LogMessage::Flush()
    @     0x7fe5b475a389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fe5b220e817  ze::cameraRigFromGflags()
    @     0x7fe5b52458b0  event_camera_simulator::DataProviderOnlineMoving3DCameraRig::DataProviderOnlineMoving3DCameraRig()
    @     0x7fe5b5238457  event_camera_simulator::loadDataProviderFromGflags()
    @     0x55af8aafe223  (unknown)
    @     0x7fe5b3bb5b97  __libc_start_main
    @     0x55af8aafed8a  (unknown)
Aborted

> Great! Closing this issue then.

I have the same issue,but  the calibration file is default that I haven't modify it.WHY?