uzh-rpg/rpg_esim #17: "Tried to insert a message with time less than ros::MIN_TIME"

francescomilano172, 2019-03-07:
Hi,
The following line throws a BagException "Tried to insert a message with time less than ros::MIN_TIME" if `time` happens to be equal to `0.5 * exposure_time_`, since the RosbagWriter tries to insert a message with timestamp exactly 0 ([https://<span></span>github.com/uzh-rpg/rpg_esim/blob/1241667c3358c90ba207fc388fb85b42bb8af5da/event_camera_simulator/esim_visualization/src/rosbag_writer.cpp#L128](https://github.com/uzh-rpg/rpg_esim/blob/1241667c3358c90ba207fc388fb85b42bb8af5da/event_camera_simulator/esim_visualization/src/rosbag_writer.cpp#L128)), which is less than 1 ns (`ros::MIN_TIME`). Simply replacing `>=` with `>` should solve the issue.

https://github.com/uzh-rpg/rpg_esim/blob/1241667c3358c90ba207fc388fb85b42bb8af5da/event_camera_simulator/esim/src/simulator.cpp#L110

Generally speaking, isn't it the case that, with the line above, `mid_exposure_time` - and therefore the timestamps of the corrupted images - can decrease even if `time` increases? In particular, if at a certain call of `publishData` `time` is equal to `a < 0.5 * exposure_time_` and at a later call `time` is `0.5 * exposure_time_ + b`, with `0 < b < a`, then the `mid_exposure_time` at the first call will be `a`, whereas the one of the second call will be `b < a`.

Thank you

supitalp, 2019-03-22:
Hi,
Thank you for this comment. Indeed, the comparison should be a strict `>` to avoid this issue.
The second issue you mention is also true (timestamps of corrupted images decreased when the exposure time has not been reached yet. I think simply avoiding to publish corrupted images when `time <= 0.5 * exposure_time` would make more sense: this way, all the corrupted images would have the correct exposure, and this would avoid this decreasing timestamp issue at initialization.
I will push a fix soon.

Thanks!
Henri

supitalp, 2019-03-22:
This should be fixed now. Closing the issue.

closed 2019-03-22 by supitalp