uzh-rpg/rpg_esim #18: Multi Objects 2D renderer not able to detect office.jpg and hence doesn't work

ashishrao7, 2019-03-14:
I have followed the instructions at this link (https://github.com/uzh-rpg/rpg_esim/wiki/Multi-Objects-2D-renderer) to render a simulation of  the multi objects 2D renderer. I have changed the absolute paths as instructed in the wiki. Below are the screenshots of that

[image]

and
[image]Pla

this is the error message I get
[image]

I can see that the files referenced by all the paths given in the config file exist. However the error message says that the package has some problem with opening office.jpg
I am able to get the Planar rendering running without any problems by only changing the absolute paths.

supitalp, 2019-03-14:
Hello,
Given the error, it is very likely that the path is wrong.
Could you please double check that the path `home/ashish/sim_ws/src/rpg_esim/rpg_esim/imp/imp_planar_renderer/textures/office.jpg` is correct?

The `--path_to_sequence_file` variable in your config file seems to point to a different folder hierarchy: `/home/ashish/sim_ws/src/rpg_esim/event_camera_simulator/...`

Thanks.

ashishrao7, 2019-03-14:
Thanks! That solved my problem. I should have looked at that part closer. I thought the later half of the address followed the default folder structure and didn't pay attention to it.

closed 2019-03-14 by ashishrao7