uzh-rpg/rpg_esim #20: cloning dependencies fatal error. Permission denied (publickey).

DrJohnDale, 2019-04-01:
When calling:
```
vcs-import < rpg_esim/dependencies.yaml
```

for each repository i get the following error like:
```
=== ./assimp_catkin (git) ===
Could not clone repository 'git@github.com:uzh-rpg/assimp_catkin.git': Cloning into '.'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

I think this is due to using the git@ style syntax.  If the command is changed to the https style everything is cloned as expected.
e.g. 
```
git@github.com:uzh-rpg/assimp_catkin.git'
```
changed to 
```
https://github.com/uzh-rpg/assimp_catkin.git'
```

Can this be changed in the dependencies.yaml in the repo? or is there another work around? 

Cheers.

supitalp, 2019-04-01:
Hi,
Do you have SSH set up correctly on Git? https://help.github.com/en/articles/adding-a-new-ssh-key-to-your-github-account

Best,
Henri

DrJohnDale, 2019-04-01:
Thanks for your help. 

I'm actually building it as part of a docker file and there is not a user as such. But ill look into if there is a way of getting around this for docker. As a short cut I am just putting the requirements into the docker file manually.

Cheers

closed 2019-04-04 by supitalp

architjaiswal, 2022-01-10:
> Hi, Do you have SSH set up correctly on Git? https://help.github.com/en/articles/adding-a-new-ssh-key-to-your-github-account
> 
> Best, Henri

I did the SSh setup as mentioned on the website but still getting the same error message and the command fails.

hypatiash, 2023-08-16:
Did you solve the problem, I syntax error when I tried to run `vcs-import < rpg_esim/dependencies.yaml`
The error comes out like this,
`Traceback (most recent call last):
  File "/usr/local/bin/vcs-import", line 11, in <module>
    load_entry_point('vcstool==0.3.0', 'console_scripts', 'vcs-import')()
  File "/home/ceren/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/ceren/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/home/ceren/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/home/ceren/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/vcstool/commands/__init__.py", line 1, in <module>
    from .branch import BranchCommand
  File "/usr/local/lib/python2.7/dist-packages/vcstool/commands/branch.py", line 6, in <module>
    from .command import Command
  File "/usr/local/lib/python2.7/dist-packages/vcstool/commands/command.py", line 6, in <module>
    from vcstool.executor import execute_jobs
  File "/usr/local/lib/python2.7/dist-packages/vcstool/executor.py", line 31
    file=stdout)
        ^
SyntaxError: invalid syntax
`

AbdessamadElK, 2024-10-09:
I had the exact same problem, cloning using the SSH link (git@ syntax) didn't work. And though cloning with the https link did work, the vcs-import command returned access denied errors.

But this solved the problem for me :
1.  Generated an SSH key and added it to my github account
2.  Cloned the repository using the SSH type link (git@github.com:uzh-rpg/rpg_esim)
3.  Executed the vcs-import command