uzh-rpg/rpg_esim #22: Is there a docker file available?

Fangyh09, 2019-04-18:
Hi, is there a docker image for it? Thanks!

supitalp, 2019-05-06:
Hi,
Currently, there is no docker image for ESIM.
I think it is a great idea but currently do not have much experience with Docker.
Did you end up making one? If yes, I would be happy to test it and maybe add it to the docker repository.
Otherwise, I may look into it in the following weeks.
I am leaving this issue open for now.

etiennedub, 2019-05-06:
Hi, 
I made a Dockerfile to use the simulator. It's really basic and some parts may need to be changed before publishing but It works for my needs.

Dockerfile :
```
FROM ros:kinetic

RUN apt-get update && \
        apt-get install -y \
        python-catkin-tools \
        python-vcstool \
        ros-kinetic-pcl-ros \
        libglfw3 libglfw3-dev \
        libglm-dev \
        ros-kinetic-hector-trajectory-server \
        git \
        ros-kinetic-opencv3 \
        autoconf \
        ros-kinetic-eigen-conversions \
        ros-kinetic-tf-conversions \
        libusb-1.0-0-dev \
        ros-kinetic-camera-info-manager \
        ros-kinetic-image-view \
        python3-catkin-pkg-modules \
        python3-pip \
        python3-tk \
        ros-kinetic-rqt* \
        ros-kinetic-rviz \
        vim \
    && apt-get autoremove \
    && apt-get clean \
    && rm -rf /tmp/* /var/lib/apt/lists/* /var/tmp/*

RUN mkdir -p /sim_ws/src
WORKDIR /sim_ws
RUN catkin config --extend /opt/ros/kinetic --cmake-args -DCMAKE_BUILD_TYPE=Release

WORKDIR /sim_ws/src
RUN git clone https://github.com/uzh-rpg/rpg_esim.git
RUN sed -i -e's/git\@github.com:/https:\/\/github.com\//g' rpg_esim/dependencies.yaml

RUN vcs-import < rpg_esim/dependencies.yaml

WORKDIR /sim_ws/src/ze_oss
RUN touch imp_3rdparty_cuda_toolkit/CATKIN_IGNORE \
      imp_app_pangolin_example/CATKIN_IGNORE \
      imp_benchmark_aligned_allocator/CATKIN_IGNORE \
      imp_bridge_pangolin/CATKIN_IGNORE \
      imp_cu_core/CATKIN_IGNORE \
      imp_cu_correspondence/CATKIN_IGNORE \
      imp_cu_imgproc/CATKIN_IGNORE \
      imp_ros_rof_denoising/CATKIN_IGNORE \
      imp_tools_cmd/CATKIN_IGNORE \
      ze_data_provider/CATKIN_IGNORE \
      ze_geometry/CATKIN_IGNORE \
      ze_imu/CATKIN_IGNORE \
      ze_trajectory_analysis/CATKIN_IGNORE
RUN /bin/bash -c 'source /ros_entrypoint.sh && catkin build esim_ros'

RUN echo "source /sim_ws/devel/setup.bash" >> ~/.bashrc

WORKDIR /
```

supitalp, 2019-07-22:
Hello,
Thanks a lot! I'm leaving the issue open for now so that people can see it. I will also add a reference to it in the wiki.

Ander-H, 2019-11-06:
> Hi,
> I made a Dockerfile to use the simulator. It's really basic and some parts may need to be changed before publishing but It works for my needs.
>
> Dockerfile :
>
> ```
> FROM ros:kinetic
>
> RUN apt-get [...]

> Hi,
> I made a Dockerfile to use the simulator. It's really basic and some parts may need to be changed before publishing but It works for my needs.
>
> Dockerfile :
>
> ```
> FROM ros:kinetic
>
> RUN apt-get [...]

Hello, I'm new to ESIM. I I intend to run ESIM by connecting to a remote server using docker. It would be a great pleasure if I can get help from you.
When you run ESIM in a container of docker, can you run the visualization (i.e., GUI)?

Ander-H, 2019-11-06:
@etiennedub 
Hello, I'm new to ESIM. I I intend to run ESIM by connecting to a remote server using docker. It would be a great pleasure if I can get help from you.
When you run ESIM in a container of docker, can you run the visualization (i.e., GUI)? 
Thank you!

and @supitalp 
Is there available docker images for ESIM? 
Thank you!

LavieC, 2019-11-06:
I have tried a lot of times to visualize esim in docker, but unfortunately,  it seems to be infeasible. Maybe you should build esim from source instead of using docker. 
If you have to use docker, @etiennedub 's dockfile works perfectly!

etiennedub, 2019-11-06:
@Ander-H Have you try to follow the instruction here : http://wiki.ros.org/docker/Tutorials/GUI ?
I can run `rqt` and other GUI tools.

Ander-H, 2019-11-11:
@etiennedub  Thanks, I will try the instruction. 

Now I have tried the VNC in docker as a remote desktop to use GUI. 
Although the GUI can work in some cases (e.g., 'rqt_image_view /dvs_rendering') when visualizing ESIM, the following error will be prompted on the bash terminal:
```shell
Qt: XKEYBOARD extension not present on the X server.
```
And in another case (i.e., 'rviz -d cfg/esim.rviz '), it can't work, and  the following error will be prompted on the bash terminal:
```shell
Qt: XKEYBOARD extension not present on the X server.
[ INFO] [1573379517.816834693]: rviz version 1.12.17
[ INFO] [1573379517.816913278]: compiled against Qt version 5.5.1
[ INFO] [1573379517.816931608]: compiled against OGRE version 1.9.0 (Ghadamon)
Segmentation fault (core dumped)
```

And @LavieC , thanks too.