uzh-rpg/rpg_esim #23: Stereo Camera Setup

DrJohnDale, 2019-05-02:
Is there an example of how to setup a stereo simulation?

I am trying to create a stereo setup but both cameras give the same information and I cannot seem to offset one from the other.

I have made a .yaml file for two cameras (see below) and tried to separate them using the T_B_C matrices, but when the simulation has finished both sets of data in the ros bag look identical.

the .yaml file is
```
label: "simulated_camera"
id: 433585ebda8d1431223927a14788a127
cameras:
- camera:
    label: dvs0
    id: a0fba5412e961934d842d3a2a78e5cba
    line-delay-nanoseconds: 0
    image_height: 180
    image_width: 240
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [198.245521288, 198.277025706, 142.064861206, 100.903484508]
    distortion:
      type: equidistant
      parameters:
        cols: 1
        rows: 4
        data: [-0.0506755889541, 0.0456313630037, -0.0825742639337, 0.0557104403236]
  T_B_C:
    cols: 4
    rows: 4
    data: [1, 0, 0,  0,
           0, -1, 0, 0,
           0,  0, -1, 0,
           0, 0, 0, 1]
- camera:
    label: dvs1
    id: a0fba5412e961934d842d3a2a78e5cbb
    line-delay-nanoseconds: 0
    image_height: 180
    image_width: 240
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [198.245521288, 198.277025706, 142.064861206, 100.903484508]
    distortion:
      type: equidistant
      parameters:
        cols: 1
        rows: 4
        data: [-0.0506755889541, 0.0456313630037, -0.0825742639337, 0.0557104403236]
  T_B_C:
      cols: 4
      rows: 4
      data: [1, 0, 0,  100.0,
             0, -1, 0, 0.0,
             0,  0, -1, 0.0,
             0, 0, 0, 1]
```

I am using a modified version of the example.conf with this .yaml file. Do I need to add any other variables to get stereo to work?

Also in the T_B_C matrix what are the units for displacement?

Thanks.

supitalp, 2019-05-06:
Hi,
The YAML file you provided does work on my side. There is no need to set up any other variable.
The baseline unit is the same as the unit of your 3D scene.
So, if you use the flying room example, you can think of it as meters. Thus, a baseline of 100 is way too large. Can you try with a smaller baseline, say 0.5 ? You should see that the images from the left and right cameras are not the same.
Also, if you open RVIZ, you should see the two camera frames of the stereo rig distinctly.

Regarding the last point (projection matrix), it is indeed correct that the shift is not present in the camera info topic. This is something we forgot to implement. Could you please file a specific issue for that problem? Thank you.

Finally, I note that your YAML file has some distortion set for each camera. Please note that distortion is not supported for the OpenGL rendering engine (the parameters will be ignored, there should be a warning message). If you really need distortion, please use the Planar or Panoramic rendering engine.

DrJohnDale, 2019-05-08:
Hi,

Thank you for the feed back. It was a bug in my code and the stereo output is working correctly. 

I have created a ticked as requested for the projection matrix (https://github.com/uzh-rpg/rpg_esim/issues/24)

Thanks for the support.

closed 2019-05-08 by DrJohnDale