uzh-rpg/rpg_esim #25: Eigen Assertion failure on OpenGL and Panoramic Renderers

silentdeadly, 2019-05-13:
After a successful `catkin build esim_ros,` I try to run `roslaunch esim_ros esim.launch config:=cfg/opengl.conf`, as per the wiki. I get the following output

`nacorti@nacorti-VirtualBox:~/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros$ roslaunch esim_ros esim.launch config:=cfg/opengl.conf
... logging to /home/nacorti/.ros/log/f613546c-752f-11e9-a031-080027fdbf14/roslaunch-nacorti-VirtualBox-30451.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nacorti-VirtualBox:34771/

SUMMARY
========

PARAMETERS
 * /hector_trajectory_server/source_frame_name: cam0
 * /hector_trajectory_server/target_frame_name: map
 * /hector_trajectory_server/trajectory_publish_rate: 15
 * /hector_trajectory_server/trajectory_update_rate: 15
 * /optic_flow_viz/arrows_scale: 0.07
 * /optic_flow_viz/arrows_step: 7
 * /optic_flow_viz/arrows_upsample_factor: 1
 * /optic_flow_viz/publish_rate: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    hector_trajectory_server (hector_trajectory_server/hector_trajectory_server)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

auto-starting new master
process[master]: started with pid [30461]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f613546c-752f-11e9-a031-080027fdbf14
process[rosout-1]: started with pid [30474]
started core service [/rosout]
process[esim_node-2]: started with pid [30489]
process[dvs_renderer-3]: started with pid [30492]
process[optic_flow_viz-4]: started with pid [30493]
process[hector_trajectory_server-5]: started with pid [30499]
esim_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 6; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
*** Aborted at 1557718448 (unix time) try "date -d @1557718448" if you are using GNU date ***
PC: @ 0xb7f16d05 ([vdso]+0xd04)
*** SIGABRT (@0x7719) received by PID 30489 (TID 0xb35baa80) from PID 30489; stack trace: ***
[ INFO] [1557718448.506023514]: Waiting for tf transform data between frames map and cam0 to become available
[esim_node-2] process has died [pid 30489, exit code -6, cmd /home/nacorti/sim_ws/devel/lib/esim_ros/esim_node --v=1 --vmodule=data_provider_from_folder=10 --flagfile=/home/nacorti/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/opengl.conf __name:=esim_node __log:=/home/nacorti/.ros/log/f613546c-752f-11e9-a031-080027fdbf14/esim_node-2.log].
log file: /home/nacorti/.ros/log/f613546c-752f-11e9-a031-080027fdbf14/esim_node-2*.log
[INFO] [1557718449.725550]: Started flow converter node
[INFO] [1557718449.733144]: Step size between arrows: 7
[INFO] [1557718449.733319]: Scale factor: 0.07
[INFO] [1557718449.733478]: Upsample factor: x1
[INFO] [1557718449.733651]: Publish rate: 100 Hz
^C[hector_trajectory_server-5] killing on exit
[optic_flow_viz-4] killing on exit
[dvs_renderer-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

The same thing happens when I try and run the panoramic renderer. However using the planar and 2d_moving_object renderers works just fine.

For reference I'm using Ubuntu 16.04 32-bit on a Virtualbox host.

OpenGL version: 3.0 Mesa 18.0.5
Python 2.7.12
GCC: version 5.4.0

I've reinstalled and gotten the same result. Is this a known issue, and is there any way to fix this? Can provide more system info on request

supitalp, 2019-05-22:
Hi,
This is not a known issue, thanks for reporting it.
Have you tried any of the steps mentioned in [that webpage](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html)? I cannot run a debugger since I cannot reproduce the crash.
I am wondering if the fact that you are running on a 32 bit system might cause the issue.
At first glance, I don't see any fundamental difference between the OpenGL and planar renderer that would explain that one works and not the other...

silentdeadly, 2019-05-22:
I haven't tried any of the steps from the Eigen.tuxfamily webpage because all of the suggestions involve code changes, which I'm reticent to make since this is all your source code.

I was wondering about the 32-bit system being an issue as well, since the Eigen issue seems to be related to bit alignment. I can try spinning up a 64-bit VM and following the same steps.

supitalp, 2019-05-23:
Yes, it is pretty likely that using a 64-bit VM will fix the issue. If you try that and it does fix the issue, please let me know, and I'll add a warning in the wiki.
Thanks!