uzh-rpg/rpg_esim #26: build error for dvs_renderer

winswang, 2019-05-22:
I am trying to 'catkin build esim_ros' but only managed to build 13 packages.
The error happens at dvs_renderer and the log is below:

CMakeFiles/dvs_renderer.dir/src/image_tracking.cpp.o: In function `dvs_renderer::ImageTracking::ImageTracking(ros::NodeHandle&)':
image_tracking.cpp:(.text+0x444): undefined reference to `image_transport::ImageTransport::ImageTransport(ros::NodeHandle const&)'
image_tracking.cpp:(.text+0x4ba): undefined reference to `image_transport::ImageTransport::advertise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool)'
image_tracking.cpp:(.text+0x58c): undefined reference to `image_transport::ImageTransport::advertise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool)'
image_tracking.cpp:(.text+0x5e6): undefined reference to `image_transport::ImageTransport::~ImageTransport()'
image_tracking.cpp:(.text+0x72b): undefined reference to `image_transport::ImageTransport::~ImageTransport()'
CMakeFiles/dvs_renderer.dir/src/image_tracking.cpp.o: In function `dvs_renderer::ImageTracking::eventsCallback(boost::shared_ptr<dvs_msgs::EventArray_<std::allocator<void> > const> const&)':
image_tracking.cpp:(.text+0x7a8): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
image_tracking.cpp:(.text+0x828): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
CMakeFiles/dvs_renderer.dir/src/image_tracking.cpp.o: In function `dvs_renderer::ImageTracking::render()':
image_tracking.cpp:(.text+0x11b2): undefined reference to `image_transport::Publisher::publish(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) const'
image_tracking.cpp:(.text+0x15e7): undefined reference to `image_transport::Publisher::publish(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) const'
CMakeFiles/dvs_renderer.dir/src/image_tracking.cpp.o: In function `dvs_renderer::ImageTracking::imageCallback(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
image_tracking.cpp:(.text+0x2071): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
image_tracking.cpp:(.text+0x2082): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
CMakeFiles/dvs_renderer.dir/src/renderer.cpp.o: In function `dvs_renderer::Renderer::~Renderer()':
renderer.cpp:(.text+0x203): undefined reference to `image_transport::Publisher::shutdown()'
renderer.cpp:(.text+0x20f): undefined reference to `image_transport::Publisher::shutdown()'
CMakeFiles/dvs_renderer.dir/src/renderer.cpp.o: In function `dvs_renderer::Renderer::imageCallback(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
renderer.cpp:(.text+0x1389): undefined reference to `image_transport::Publisher::publish(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) const'
CMakeFiles/dvs_renderer.dir/src/renderer.cpp.o: In function `dvs_renderer::Renderer::eventsCallback(boost::shared_ptr<dvs_msgs::EventArray_<std::allocator<void> > const> const&)':
renderer.cpp:(.text+0x19b0): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
renderer.cpp:(.text+0x214d): undefined reference to `image_transport::Publisher::publish(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) const'
renderer.cpp:(.text+0x2189): undefined reference to `image_transport::Publisher::getNumSubscribers() const'
renderer.cpp:(.text+0x27b4): undefined reference to `image_transport::Publisher::publish(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&) const'
CMakeFiles/dvs_renderer.dir/src/renderer.cpp.o: In function `dvs_renderer::Renderer::Renderer(ros::NodeHandle&, ros::NodeHandle)':
renderer.cpp:(.text+0x36c3): undefined reference to `image_transport::ImageTransport::ImageTransport(ros::NodeHandle const&)'
renderer.cpp:(.text+0x3b6a): undefined reference to `image_transport::ImageTransport::subscribe(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, boost::function<void (boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)> const&, boost::shared_ptr<void> const&, image_transport::TransportHints const&)'
renderer.cpp:(.text+0x3d8d): undefined reference to `image_transport::ImageTransport::advertise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool)'
renderer.cpp:(.text+0x3e19): undefined reference to `image_transport::ImageTransport::advertise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool)'
renderer.cpp:(.text+0x4111): undefined reference to `image_transport::ImageTransport::~ImageTransport()'
renderer.cpp:(.text+0x434d): undefined reference to `image_transport::ImageTransport::~ImageTransport()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/winston/esim_ws/devel/.private/dvs_renderer/lib/dvs_renderer/dvs_renderer] Error 1
make[1]: *** [CMakeFiles/dvs_renderer.dir/all] Error 2
make: *** [all] Error 2

supitalp, 2019-05-22:
Hello,
The error happens in the `dvs_renderer` package, which is not part of ESIM, but rather the [ROS driver](https://github.com/uzh-rpg/rpg_dvs_ros). Could you please file an issue there?
Please also indicate what OS / version of ROS you are using.
Thanks.

winswang, 2019-05-22:
> Hello,
> The error happens in the `dvs_renderer` package, which is not part of ESIM, but rather the [ROS driver](https://github.com/uzh-rpg/rpg_dvs_ros). Could you please file an issue there?
> Please also indicate what OS / version of ROS you are using.
> Thanks.

Thank you! I followed the dvs_renderer instructions and was able to build it. 

Then I'm rebuilding the 'esim_ros' and my progress has come to 31/36 complete. But I'm still stuck at 'assimp_catkin', where I only see the timer counting forever. I don't know what's happening...

supitalp, 2019-05-22:
Good. Could you please let me know what specific instructions fixed the building issue for the `dvs_renderer` package, so that I can add it to the wiki here?

`assimp` is a pretty large library, it may take some time to compile, please wait a bit more :)

winswang, 2019-05-22:
I replaced `catkin_simple` and `rpg_dvs_ros`. And then catkin build `dvs_ros_driver`. I got into a similar problem where `libcaer_catkin` takes too long time. I then killed it and skipped to `cd libcaer_catkin` and `sudo ./install.sh`. And then `catkin build dvs_renderer`. 

> All 3 packages succeeded!

tstellanova, 2019-06-06:
Encountering a similar problem: Unable to build esim_ros because apparently `dvs_renderer` has an unmet dependency on `image_transport`. 

Would it be possible to pin `rpg_esim`'s dependency on `dvs_renderer` to a particular git commit that is known to work?

closed 2019-07-12 by winswang