uzh-rpg/rpg_esim #29: how to change the camera resolution and turn on color option

winswang, 2019-07-12:
Can someone point me to the specific location where I can change the camera parameter? Or the relevant tutorial/demo having the option to do that? 
I'm primarily playing with the esim_ros for the image plane demo in the wiki.

I'm also wondering how to turn on the color option to output color frames.

Much appreciated!

supitalp, 2019-07-15:
Hello,

1. How to change the camera intrinsic parameters? (field of view, etc.)

These parameters are specified in the camera calibration file specified in the config file, e.g. [this file](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/cfg/calib/example.conf#L14). The entries of the `data` line are `[fx, fy, cx, cy]`, where `fx`, `fy` are the horizontal (resp. vertical) focal lengths, and `cx`, `cy` the horizontal (resp. vertical) position of the optical center. Everything expressed in pixels.

2. How to turn on the color option?

Please switch to [this branch](https://github.com/uzh-rpg/rpg_esim/tree/feature/color) and rebuild:

```
git fetch
git checkout feature/color
catkin build esim_ros
```

This will activate color support in ESIM, e.g. the images published will now be color images.
To activate color event simulation (as in [our paper](http://rpg.ifi.uzh.ch/CED.html)), you need to add the following line to your config file:

```--simulate_color_events=True```

An example config file for color is available [here](https://github.com/uzh-rpg/rpg_esim/blob/feature/color/event_camera_simulator/esim_ros/cfg/example_color.cfg#L14).

Hope that helps!

winswang, 2019-07-16:
Hi, thanks for your reply. I have checked out the feature/color branch but failed to build it.
The error is:

> Errors     << esim_ros:make /mnt/f/Winston/sim_ws/logs/esim_ros/build.make.003.log
make[2]: *** No rule to make target '/mnt/f/Winston/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg', needed by '/mnt/f/Winston/sim_ws/devel/.private/esim_ros/include/esim_ros/example_colorConfig.h'.  Stop.
make[1]: *** [CMakeFiles/esim_ros_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /mnt/f/Winston/sim_ws/build/esim_ros; catkin build --get-env esim_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

There is also one warning before the build error in case helpful:

> Warnings   << esim_data_provider:make /mnt/f/Winston/sim_ws/logs/esim_data_provider/build.make.003.log
In file included from /mnt/f/Winston/sim_ws/devel/.private/glog_catkin/include/glog/logging.h:86:0,
                 from /mnt/f/Winston/sim_ws/src/ze_oss/ze_common/include/ze/common/logging.hpp:32,
                 from /mnt/f/Winston/sim_ws/src/rpg_esim/event_camera_simulator/esim_data_provider/src/data_provider_factory.cpp:1:
/mnt/f/Winston/sim_ws/src/rpg_esim/event_camera_simulator/esim_data_provider/src/data_provider_factory.cpp:28:8: warning: unknown escape sequence: '\040'
        "...based on optic flow: deltaT = lambda \ max(||du/dt||) where du/dt denotes the 2D optic flow field.");
        ^
/mnt/f/Winston/sim_ws/devel/.private/gflags_catkin/include/gflags/gflags.h:462:34: note: in definition of macro ‘MAYBE_STRIPPED_HELP’
 #define MAYBE_STRIPPED_HELP(txt) txt
                                  ^
/mnt/f/Winston/sim_ws/devel/.private/gflags_catkin/include/gflags/gflags.h:531:4: note: in expansion of macro ‘DEFINE_VARIABLE’
    DEFINE_VARIABLE(double, D, name, val, txt)
    ^
/mnt/f/Winston/sim_ws/src/rpg_esim/event_camera_simulator/esim_data_provider/src/data_provider_factory.cpp:24:1: note: in expansion of macro ‘DEFINE_double’
 DEFINE_double(simulation_adaptive_sampling_lambda, 0.5,
 ^
cd /mnt/f/Winston/sim_ws/build/esim_data_provider; catkin build --get-env esim_data_provider | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

Do you know how to fix this? I'm using Windows 10 with subsystem for Ubuntu16.04 (app). I was able to build and run data before checking out the `feature/color` branch.

supitalp, 2019-07-16:
Hi,
These errors do not make much sense to me. Could you try recompiling ESIM from scratch?
You can force that by removing the build folders before rebuilding:

```bash
# remove build folders
roscd
cd ../
rm -rf build/esim_* devel/lib/libesim_*
# recompile
catkin build esim_ros
```

Please let me know if that helps.

winswang, 2019-07-16:
Thank you for the timely reply. I was able to recompile and run the planar demo. 

I have a remaining question: the planar demo does not stop, and every time I have to manually stop the process. I notice there are something remain yet to publish such as optic flow. The output after I ctrl+c is like this (otherwise it will not proceed/quit):

> I0716 18:22:55.754420 26164 rosbag_writer.cpp:62] Finalizing the bag...
I0716 18:22:56.010391 26164 rosbag_writer.cpp:64] Finished writing to bag: /mnt/f/Winston/data/esim-output/bags/color-out.bag
[esim_node-2] process has finished cleanly
log file: /home/winswang/.ros/log/7fb0a0f2-a7b3-11e9-a151-4cedfb3ce9d2/esim_node-2*.log
^C[hector_trajectory_server-5] killing on exit
[optic_flow_viz-4] killing on exit
[dvs_renderer-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

How do I automatically make it quit. And how can I publish the optical flow? Is there a demo?

supitalp, 2019-07-16:
Hi,

### 1. How to automatically quit ESIM?

It is not possible to quit automatically when launching with `roslaunch`. However, it is possible to do so by launching the ESIM node directly, as follows (make sure to have a running `roscore` first):

```bash
roscd esim_ros
rosrun esim_ros esim_node --flagfile=cfg/example_color.cfg
```

The process will quit upon reaching the end of the camera trajectory.

### 2. How to publish optic flow maps?

The parameter `--ros_publisher_optic_flow_rate` (see [here](https://github.com/uzh-rpg/rpg_esim/blob/feature/color/event_camera_simulator/esim_ros/cfg/example_color.cfg#L53)) specifies the output framerate of the optic flow maps (set it to a value > 0, e.g. 30 Hz).

To visualize the flow maps, launch ESIM:

```bash
roslaunch esim_ros esim.launch config:=cfg/example_color.cfg
```

and visualize the topic `/flow_color` or `/flow_arrow` in `rqt_image_view` for example.

Hope that helps.

winswang, 2019-07-16:
Hi, thanks for the information. How do I make sure I have a running roscore?

supitalp, 2019-07-16:
Hi,
You can open a separate terminal, and run the following command:

```roscore```

If you are not too familiar with ROS, it could be helpful to have a look at the [ROS tutorials](http://wiki.ros.org/ROS/Tutorials) :)

winswang, 2019-07-16:
Thank you! It's very helpful!

realArtichoke, 2020-01-28:
Hi, I am trying to install feature/color branch and I have trouble with building.
```
Errors     << esim_ros:make /home/imagingtre/catkin_ws/logs/esim_ros/build.make.014.log
/home/imagingtre/catkin_ws/build/esim_ros/setup_custom_pythonpath.sh: 5: exec: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: Permission denied
make[2]: *** [/home/imagingtre/catkin_ws/devel/.private/esim_ros/include/esim_ros/example_colorConfig.h] Error 126
make[1]: *** [CMakeFiles/esim_ros_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/imagingtre/catkin_ws/build/esim_ros; catkin build --get-env esim_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << esim_ros:make                        [ Exited with code 2 ]      
Failed    <<< esim_ros                             [ 0.6 seconds ]             
[build] Summary: 35 of 36 packages succeeded.                                  
[build]   Ignored:   21 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 11.2 seconds total.                         
```

In similar issues I've seen an advice to execute  chmod +x for that cfg file. But then I get this whilst building:

```
Errors     << esim_ros:make /home/imagingtre/catkin_ws/logs/esim_ros/build.make.015.log
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 1: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --vmodule=data_provider_online_render=0,camera_simulator=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 2: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --random_seed=50: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 3: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --data_source=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 4: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --path_to_output_bag=/tmp/out.bag: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 6: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_pos=0.15: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 7: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_neg=0.15: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 8: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_sigma_pos=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 9: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_sigma_neg=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 11: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --exposure_time_ms=20.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 12: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --use_log_image=true: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 13: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --log_eps=0.001: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 14: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --simulate_color_events=true: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 16: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --calib_filename=/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/calib/pinhole_mono_nodistort.yaml: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 18: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_type=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 19: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_texture=/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/imp/imp_planar_renderer/textures/carpet.jpg: not found
[... cut ...]
Failed    <<< esim_ros                             [ 0.6 seconds ]            
[build] Summary: 35 of 36 packages succeeded.                                 
[build]   Ignored:   21 packages were skipped or are blacklisted.             
[build]   Warnings:  None.                                                    
[build]   Abandoned: None.                                                    
[build]   Failed:    1 packages failed.                                       
[build] Runtime: 10.2 seconds total.          
```
I will be very grateful for any help with that.

dnzrgn, 2020-04-30:
> Hi,
>
>1. How to automatically quit ESIM?
>It is not possible to quit automatically when launching with roslaunch. However, it is possible to do >so by launching the ESIM node directly, as follows (make sure to have a running roscore first):

It works for me when adding the 'required="true"' parameter to the node definition in the 'esim.launch' file:

```
<launch>

  <arg name="config" />

  <!-- Event camera simulator -->
  <node name="esim_node" pkg="esim_ros" type="esim_node" args="
    --v=1
    --vmodule=data_provider_from_folder=10
    --flagfile=$(find esim_ros)/$(arg config)
    " output="screen" required="true" />

  <include file="$(find esim_ros)/launch/visualization.launch" />

</launch>
```

I don't know if it's the way to go. Just wanted to mention it, because it really simplified my workflow and I had to spend some time finding it out.

ezragblock, 2024-04-18:
> Hi, I am trying to install feature/color branch and I have trouble with building.
>
> ```
> Errors     << esim_ros:make /home/imagingtre/catkin_ws/logs/esim_ros/build.make.014.log [...]

Did you end up solving this issue? I'm having the same issue.