uzh-rpg/rpg_esim #31: which version of opencv did you use?

yuanhaowang1213, 2019-07-18:
I got an error: 

rpg_dvs_ros/dvs_renderer/src/renderer.cpp:193:11: error: ‘undistort’ is not a member of ‘cv’

when build the esim_ros. So which version of opencv did you use? Or how to resolve it? I  am using opencv 4.0.

supitalp, 2019-07-19:
Hi,
The code was tested with OpenCV 3.2 (which is the version of OpenCV shipped with ROS Melodic).
It is possible that OpenCV 4.0 changed the API. What version of ROS are you using?

Note that this is a problem with the `dvs_renderer` package, so may file the issue on the `rpg_dvs_ros` repo directly: https://github.com/uzh-rpg/rpg_dvs_ros/issues

ESIM does not strictly require this package, it is only used for visualization of the events.

supitalp, 2019-07-19:
That being said, it looks like you could solve the issue by adding the following include after the includes in the file: https://github.com/uzh-rpg/rpg_dvs_ros/blob/master/dvs_renderer/src/renderer.cpp#L4

```#include <opencv2/calib3d/calib3d.hpp>```

Please let me know if that solves the problem.

yuanhaowang1213, 2019-07-19:
Another error:
src/ze_oss/imp_bridge_opencv/include/imp/bridge/opencv/cv_bridge.hpp:46:32: error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
     mat = cv::imread(filename, CV_LOAD_IMAGE_GRAYSCALE);
src/ze_oss/imp_bridge_opencv/include/imp/bridge/opencv/cv_bridge.hpp:52:32: error: ‘CV_LOAD_IMAGE_COLOR’ was not declared in this scope
     mat = cv::imread(filename, CV_LOAD_IMAGE_COLOR)

Anyway, I will change to opencv 3.2 to have a try.

supitalp, 2019-07-19:
Thanks for reporting. Switching to OpenCV 3.2 should fix your issues.
However, it seems like it should not be too difficult to fix these small issues with OpenCV 4 since they seem to stem from small API / name changes. If you do so, a pull request with those fixes for OpenCV 4 would be much appreciated :)

yuanhaowang1213, 2019-07-19:
Hi, there are several name need to be changes for opencv 4.0. Such as cv_MINMAX need to be changed to cv::NORM_MINMAX, et al. 

But I have another question.

My devel folder and src folder are in the home directory.

[image]

The folder list of my workstation is:
build           Downloads                 Pictures          src
CMakeLists.txt  examples.desktop          Public            Templates
Desktop         logs                      setupeventsim.sh  Videos
devel           Music                     sim_ws
Documents       NVIDIA_CUDA-10.0_Samples  snap

After the installation, I tried use the source ~/devel/setup.bash, but nothing happens. There is no start of esim. Do I need to change some bash file? 

Thank you

supitalp, 2019-07-22:
Hello,
Please follow the instructions on the wiki to launch ESIM. For example: https://github.com/uzh-rpg/rpg_esim/wiki/3D-OpenGL-Rendering-Engine

Sourcing `~/devel/setup.bash` does **not** launch ESIM, it just sources the ROS environment into the current terminal.

closed 2019-07-24 by yuanhaowang1213