uzh-rpg/rpg_esim #33: Different trajectory in output rosbag to the input csv file

yunxdai, 2019-07-20:
Hello! I tried to use "csv input" mode of ESIM to input specific trajectories, while I found sometimes the actual trajectories didn't fit well to the input. Then I carried out several simple experiments on it.

The trajectory I generated drove the object moving on only x-axis in constant velocity, keeping y and z in 0. It turned out that the actual path was bad. Considering the usage of spline interpolation, I generated a acceleration on x axis and the result seems quite great ( showed in the following figures)
[image]

Then I created a circular motion trajectory on x-y plane in constant angular velocity and the output fitted the expected path well too. ( traj, x, y comparison showed in the following figures)
[image]
[image]
[image]

However, the problem occured again when I attempted to input a spiral motion in 3 dimension ( see in the first following figure). The output path appeared as severe shrink comparing to the expected trajectory. And it was strange that motion on x and y axis were shrinking, while the motion on z axis seems acceptable. In this case, object followed circular motion on x-y plane, and acceleration on z axis. (details in the figures)
[image]
[image]
[image]
[image]
I couldn't figure out why it happened and how I should fixed this bug, so would you mind give me suggestions on it?
**Notes:**
-  I got the output trajectories from the output rosbag file, whose path was defined in the configuration files.
-  All of my attempts were under opengl renderers.
-  I changed the spline segments number to 200 as your recommendation in the previous similar issue. Meanwhile, I also set all the "multipliers" to 1 although I didn't know their exact function in simulator. I would appreciate it very much if you could tell their roles in the simulator.

Thank you in advance for your help!

supitalp, 2019-07-22:
Hello,
Thanks for the detailed report.
Can you try changing the parameter `--trajectory_lambda` to `0` (or a very small value)?
This parameter limits the maximum acceleration along the trajectory (to guarantee that the trajectory is somewhat smooth), so it could explain why your trajectory seems to "underfit" the reference one in this case.

Please let me know if that helps.

yunxdai, 2019-07-22:
Thank you for your reply! It did work after changing that parameter, and I am curious how this parameter represent the limitation. Is its value exactly equal to the maximum acceleration in every dimension, or just a relative value? I think this limitation is important in simulation so I want to confirm its exact mean rather than just changing it to 0. So it will be quite helpful if you can provide more information about it.