uzh-rpg/rpg_esim #34: unable to install (error happened with catkin build esim_ros)

wl082013, 2019-07-23:
H, I am installing ESIM, but no matter how I tried, there always comes with error  when I start 'catkin build esim_ros'. Could you give some help on  why this happened? 
Besides, I wonder what are the requirements of python version, cmake and gcc versions ?  Aso is it possible to use conda python instead of pip ?  it seems esim is contradicting to conda. 
Thanks four your help.

supitalp, 2019-07-23:
Hello,
To help you I would need more details on your setup (OS, cmake, gcc versions, etc.) and the compilation log.
ESIM depends on ROS, which requires Python 2, so it may conflict with conda indeed.

wl082013, 2019-07-23:
Hi, my PC is installed with Ubuntu 16.04, cmake version 3.5.1 and gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11). 
I followed the installation, but always fail when starting 'catkin build esim_ros'.

supitalp, 2019-07-23:
Hello,
The log you included is not enough. Something seems to be failing before the lines you included. Could you please send the full log?

wl082013, 2019-07-23:
Hi it seems there are many log files in the sim_ws, which one is more relevant? 
[image]
thank you for your help

supitalp, 2019-07-23:
Hi,
I am not sure whether the catkin build logs are written to disk or not.
The screenshot you sent was good, however I just need all of the output lines since running the `catkin build esim_ros` command. Can you copy paste those from the terminal to here?

wl082013, 2019-07-23:
-------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/addisonwang/sim_ws
-------------------------------------------------------------
Build Space:        [exists] /home/addisonwang/sim_ws/build
Devel Space:        [exists] /home/addisonwang/sim_ws/devel
Install Space:      [unused] /home/addisonwang/sim_ws/install
Log Space:          [exists] /home/addisonwang/sim_ws/logs
Source Space:       [exists] /home/addisonwang/sim_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------
[build] Found '44' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Finished  <<< catkin_simple                        [ 0.1 seconds ]             
Starting  >>> assimp_catkin                                                    
Starting  >>> dvs_msgs                                                         
Starting  >>> eigen_catkin                                                     
Starting  >>> esim_msgs                                                        
Starting  >>> gflags_catkin                                                    
Starting  >>> yaml_cpp_catkin                                                  
Starting  >>> ze_cmake                                                         
Finished  <<< ze_cmake                             [ 0.1 seconds ]             
Finished  <<< esim_msgs                            [ 0.2 seconds ]             
Finished  <<< dvs_msgs                             [ 0.2 seconds ]             
Starting  >>> dvs_renderer                                                     
Finished  <<< yaml_cpp_catkin                      [ 0.1 seconds ]             
Finished  <<< eigen_catkin                         [ 0.1 seconds ]             
Finished  <<< gflags_catkin                        [ 0.1 seconds ]             
Starting  >>> glog_catkin                                                      
Finished  <<< assimp_catkin                        [ 0.1 seconds ]             
_______________________________________________________________________________
Errors     << glog_catkin:make /home/addisonwang/sim_ws/logs/glog_catkin/build.make.002.log
/bin/sh: 1: autoreconf: not found
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-configure] Error 127
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/addisonwang/sim_ws/build/glog_catkin; catkin build --get-env glog_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << glog_catkin:make                     [ Exited with code 2 ]      
Failed    <<< glog_catkin                          [ 0.1 seconds ]             
Abandoned <<< eigen_checks                         [ Unrelated job failed ]    
Abandoned <<< esim_unrealcv_bridge                 [ Unrelated job failed ]    
Abandoned <<< minkindr                             [ Unrelated job failed ]    
Abandoned <<< minkindr_conversions                 [ Unrelated job failed ]    
Abandoned <<< ze_common                            [ Unrelated job failed ]    
Abandoned <<< imp_core                             [ Unrelated job failed ]    
Abandoned <<< imp_bridge_opencv                    [ Unrelated job failed ]    
Abandoned <<< imp_bridge_ros                       [ Unrelated job failed ]    
Abandoned <<< ze_cameras                           [ Unrelated job failed ]    
Abandoned <<< esim_common                          [ Unrelated job failed ]    
Abandoned <<< esim_rendering                       [ Unrelated job failed ]    
Abandoned <<< esim_visualization                   [ Unrelated job failed ]    
Abandoned <<< imp_multi_objects_2d                 [ Unrelated job failed ]    
Abandoned <<< imp_opengl_renderer                  [ Unrelated job failed ]    
Abandoned <<< imp_panorama_renderer                [ Unrelated job failed ]    
Abandoned <<< imp_planar_renderer                  [ Unrelated job failed ]    
Abandoned <<< imp_unrealcv_renderer                [ Unrelated job failed ]    
Abandoned <<< ze_matplotlib                        [ Unrelated job failed ]    
Abandoned <<< ze_ros                               [ Unrelated job failed ]    
Abandoned <<< ze_visualization                     [ Unrelated job failed ]    
Abandoned <<< ze_splines                           [ Unrelated job failed ]    
Abandoned <<< ze_vi_simulation                     [ Unrelated job failed ]    
Abandoned <<< esim_trajectory                      [ Unrelated job failed ]    
[... cut ...]
[build] Summary: 9 of 36 packages succeeded.                                   
[build]   Ignored:   8 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 26 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1.0 seconds total.                                            
(base) addisonwang@addisonwang-H370HD3:~/sim_ws/src/ze_oss$

supitalp, 2019-07-23:
OK, now I can see the error message in the `glog_catkin` package: ` /bin/sh: 1: autoreconf: not found`.
Installing `autoconf should fix the problem:

```sudo apt-get install autoconf```

wl082013, 2019-07-23:
okay thanks I will try

wl082013, 2019-07-23:
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/addisonwang/sim_ws
-------------------------------------------------------------
Build Space:        [exists] /home/addisonwang/sim_ws/build
Devel Space:        [exists] /home/addisonwang/sim_ws/devel
Install Space:      [unused] /home/addisonwang/sim_ws/install
Log Space:          [exists] /home/addisonwang/sim_ws/logs
Source Space:       [exists] /home/addisonwang/sim_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------
[build] Found '44' packages in 0.0 seconds.                                                                                       
[build] Package table is up to date.                                                                                              
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                                               
__________________________________________________________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/addisonwang/sim_ws/logs/catkin_tools_prebuild/build.cmake.004.log                 
ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/addisonwang/anaconda3/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/addisonwang/sim_ws/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:167 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/addisonwang/sim_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/addisonwang/sim_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/addisonwang/sim_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/addisonwang/sim_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
..................................................................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]                                                         
Failed    <<< catkin_tools_prebuild                [ 0.2 seconds ]                                                                
Abandoned <<< catkin_simple                        [ Unrelated job failed ]                                                       
Abandoned <<< assimp_catkin                        [ Unrelated job failed ]                                                       
Abandoned <<< dvs_msgs                             [ Unrelated job failed ]                                                       
Abandoned <<< dvs_renderer                         [ Unrelated job failed ]                                                       
Abandoned <<< eigen_catkin                         [ Unrelated job failed ]                                                       
Abandoned <<< esim_msgs                            [ Unrelated job failed ]                                                       
Abandoned <<< gflags_catkin                        [ Unrelated job failed ]                                                       
Abandoned <<< glog_catkin                          [ Unrelated job failed ]                                                       
Abandoned <<< eigen_checks                         [ Unrelated job failed ]                                                       
Abandoned <<< esim_unrealcv_bridge                 [ Unrelated job failed ]                                                       
Abandoned <<< minkindr                             [ Unrelated job failed ]                                                       
Abandoned <<< minkindr_conversions                 [ Unrelated job failed ]                                                       
Abandoned <<< yaml_cpp_catkin                      [ Unrelated job failed ]                                                       
Abandoned <<< ze_cmake                             [ Unrelated job failed ]                                                       
[... cut ...]
[build]   Ignored:   8 packages were skipped or are blacklisted.                                                                  
[build]   Warnings:  None.                                                                                                        
[build]   Abandoned: 36 packages were abandoned.                                                                                  
[build]   Failed:    1 packages failed.                                                                                           
[build] Runtime: 0.4 seconds total.

wl082013, 2019-07-23:
There is still error happening. I installed catkin_pkg, but it shows there is no catkin_pkg.

supitalp, 2019-07-23:
Hello,
A quick Google search with the error message ` ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'` leads to [this page](https://afsyaw.wordpress.com/2016/12/23/setting-up-ros-on-ubuntu-16-04/), which suggests that installing outside of a conda environment might be a solution.
Can you please try that?

wl082013, 2019-07-23:
okay, I did not uninstall conda, I guess that's the problem.

supitalp, 2019-07-24:
You do not need to uninstall conda. It should be enough to comment out Anaconda's path in your `~/.bashrc` before launching the installation.

closed 2019-08-14 by wl082013