uzh-rpg/rpg_esim #35: How to save data

eugenelyj, 2019-07-23:
Thanks for your sharing!  But how could i save the data from the simulator?

closed 2019-07-23 by eugenelyj

supitalp, 2019-07-23:
Hi,
## Simple solution
The data will be saved as a [rosbag](http://wiki.ros.org/rosbag).
You can specify the output location of the rosbag with the parameter `--path_to_output_bag`.

## Custom data format (advanced)
In case you would prefer to save the data in your custom format, you may write your own publisher implementing the [publisher interface](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_visualization/include/esim/visualization/publisher_interface.hpp), and add it to the list of publishers (example [here](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/src/esim_node.cpp#L78)).
Of course this requires to write some C++ code and recompile.

Hope that helps!

eugenelyj, 2019-07-23:
Yes, it helps! Thank you so much.