uzh-rpg/rpg_esim #36: rviz no output

yuanhaowang1213, 2019-08-05:
I have tried the Multi Objects 2d renderer example, and run 
roslaunch esim_ros esim.launch config:=cfg/multi_objects.conf

the out put is: 
... logging to /home/wangy0k/.ros/log/99a41850-b74d-11e9-a932-ac1f6bd0c91d/roslaunch-VCC-SYS-7039A-I-4713.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://VCC-SYS-7039A-I:39191/

SUMMARY
========

PARAMETERS
 * /hector_trajectory_server/source_frame_name: cam0
 * /hector_trajectory_server/target_frame_name: map
 * /hector_trajectory_server/trajectory_publish_rate: 15
 * /hector_trajectory_server/trajectory_update_rate: 15
 * /optic_flow_viz/arrows_scale: 0.07
 * /optic_flow_viz/arrows_step: 7
 * /optic_flow_viz/arrows_upsample_factor: 1
 * /optic_flow_viz/publish_rate: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    hector_trajectory_server (hector_trajectory_server/hector_trajectory_server)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

auto-starting new master
process[master]: started with pid [4723]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 99a41850-b74d-11e9-a932-ac1f6bd0c91d
process[rosout-1]: started with pid [4736]
started core service [/rosout]
process[esim_node-2]: started with pid [4747]
process[dvs_renderer-3]: started with pid [4754]
process[optic_flow_viz-4]: started with pid [4755]
process[hector_trajectory_server-5]: started with pid [4756]
I0805 09:52:33.680042  4747 imp_multi_objects_2d_renderer.cpp:44] width = 346, height = 260
I0805 09:52:33.680193  4747 imp_multi_objects_2d_renderer.cpp:45] tmax = 1
I0805 09:52:33.680214  4747 object.cpp:12] Loading texture file from file: /home/wangy0k/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_planar_renderer/textures/office.jpg.
[ INFO] [1564987953.684704422]: Waiting for tf transform data between frames map and cam0 to become available
I0805 09:52:33.791405  4747 object.cpp:24] Pre-filtering the texture with median filter of size: 15.
[INFO] [1564987954.093177]: Started flow converter node
[INFO] [1564987954.093433]: Step size between arrows: 7
[INFO] [1564987954.093572]: Scale factor: 0.07
[INFO] [1564987954.093796]: Upsample factor: x1
[INFO] [1564987954.093885]: Publish rate: 100 Hz
I0805 09:52:35.000685  4747 object.cpp:31] Pre-filtering the texture with gaussian filter of size: 1.
I0805 09:52:35.241488  4747 object.cpp:12] Loading texture file from file: /home/wangy0k/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_multi_objects_2d/textures/rugby_ball.png.
I0805 09:52:35.260526  4747 object.cpp:24] Pre-filtering the texture with median filter of size: 11.
I0805 09:52:35.379128  4747 object.cpp:31] Pre-filtering the texture with gaussian filter of size: 1.
I0805 09:52:35.405050  4747 ros_publisher.cpp:37] Initializing ROS
I0805 09:52:35.511757  4747 rosbag_writer.cpp:39] Will write to bag: /tmp/out.bag
W0805 09:52:35.511781  4747 adaptive_sampling_benchmark_publisher.cpp:57] Empty benchmark folder string: will not write benchmark files
W0805 09:52:35.511885  4747 synthetic_optic_flow_publisher.cpp:41] Empty output folder string: will not write synthetic optic flow files
I0805 09:52:35.518921  4747 event_simulator.cpp:27] Converting the image to log image with eps = 0.001.
I0805 09:52:35.519232  4747 event_simulator.cpp:11] Initialized event camera simulator with sensor size: [346 x 260]
I0805 09:52:35.519243  4747 event_simulator.cpp:12] and contrast thresholds: C+ = 0.5 , C- = 0.5
W0805 09:52:35.520195  4747 camera_simulator.cpp:43] The images do not cover a time span long enough to simulate the exposure time accurately.
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
I0805 09:52:48.169306  4747 rosbag_writer.cpp:62] Finalizing the bag...
I0805 09:52:48.172581  4747 rosbag_writer.cpp:64] Finished writing to bag: /tmp/out.bag
[esim_node-2] process has finished cleanly
log file: /home/wangy0k/.ros/log/99a41850-b74d-11e9-a932-ac1f6bd0c91d/esim_node-2*.log
[ WARN] [1564987973.686946297]: No transform between frames map and cam0 available after 20.002202 seconds of waiting. This warning only prints once.

And after that, I run the rviz 
roscd esim_visualization
rviz -d cfg/esim.rviz

It shows no image. It seems the bag file is not read.

supitalp, 2019-08-05:
Hello,
The output looks normal, no errors there. Please let me clarify what ESIM is actually doing.

- Upon launching the ESIM node (`roslaunch ...`), ESIM starts publishing simulated readings on different topics (`/cam0/events`, etc.).

- If you happen to have some visualization tool launched (e.g. `rviz` or `rqt`), some visualizations of the sensors will show while the simulation is running (note that the `multi_objects_2d` rendering does not output any camera trajectory of 3D map in RVIZ as this is only a 2D rendering engine -- in this case, better use the `rqt` visualization tool).

- Note that in parallel to publishing the simulated readings on ROS topics, ESIM also saves all the data in `rosbag` (hence the message: `Finished writing to bag: /tmp/out.bag`), for later usage.

So, it you want to replay one simulation, you can:

1. Make sure you have a visualization tool open (e.g. `rviz` or `rqt`, or both).
2. Play the recorded rosbag as follows: `rosbag play /path/to/saved_bag.bag`). Then, some visualization should be visible.

Hope that helps.

yuanhaowang1213, 2019-08-05:
Thank you. Now I understand.
I have tried rosbag play /tmp/out.bag And there is an error: 
[ERROR] [1564989395.828164974]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

I have tried 
export | grep ROS
the output:
declare -x ROSLISP_PACKAGE_DIRECTORIES=""
declare -x ROS_DISTRO="kinetic"
declare -x ROS_ETC_DIR="/opt/ros/kinetic/etc/ros"
declare -x ROS_MASTER_URI="http://localhost:11311"
declare -x ROS_PACKAGE_PATH="/opt/ros/kinetic/share"
declare -x ROS_ROOT="/opt/ros/kinetic/share/ros"
declare -x ROS_VERSION="1"

I think it might be a setting error for ros. I am quite new to this. And I will try to fix it. 

Thank you.

supitalp, 2019-08-16:
Hello,
You need to have a roscore running to play a rosbag.
You can launch a roscore by opening a new terminal, and running: `roscore`.
Then, you should be able to play the rosbag.

closed 2019-09-10 by supitalp