uzh-rpg/rpg_esim #37: how to simulate motion blur?

eugenelyj, 2019-08-07:
Thanks for your amazing work!!
I am not sure that how do esim simulate the motion blur？Does it use some explosure simulation then when camera move fast, it would bring motion blur while camera move slowly it would not. Or does it just use some blur kernel like guassian kernel to process the image?

eugenelyj, 2019-08-07:
Another question, can i publish blurred frame and non-blurred frame at the same time?

supitalp, 2019-08-07:
Hello,

> Does it use some explosure simulation then when camera move fast, it would bring motion blur while camera move slowly it would not. Or does it just use some blur kernel like guassian kernel to process the image?

ESIM does proper exposure simulation based on integration (see Section 7.4 in [our paper](http://rpg.ifi.uzh.ch/docs/CORL18_Rebecq.pdf)). So, yes, the amount of gaussian blur depends on the relative speed between the camera and the environment.
Note that you can set the exposure time (in ms) by changing the flag `--exposure_time_ms` in your config file, e.g.: https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/cfg/example.conf#L11

> Another question, can i publish blurred frame and non-blurred frame at the same time?

Yes, both are published at the same time on the topics `/cam0/image_raw` and `/cam0/image_corrupted`.

eugenelyj, 2019-08-07:
@supitalp Thank you so much!! It helps.

closed 2019-08-07 by eugenelyj

eugenelyj, 2019-08-26:
@supitalp Hi!
 I want to simulate fast/slow camera motion, which causes heavy/not-heavy motion blur, how can i do that? Can i control the camera moving speed directly, in the mode `Random spline trajectory`? i just do not find that option, but i think it is a trick way to change exposure time instead.