uzh-rpg/rpg_esim #38: Change position of 3D model

yunxdai, 2019-08-07:
Hello! I am trying to use 3D opengl render of ESIM, but I did not find out how to change the position of input 3D model. My task required another another 3D scene for testing and I already had the 3D model. But when I just changed the file path parameter in config file, my 3D model was not fixed in a proper place. Therefore I want to figure out whether there is any way to set the position of 3D model?
Thank you in advance for your help!

supitalp, 2019-08-16:
Hello,
It is currently not possible to explicitly set the position of the 3D model, unfortunately.
You could either:
  1. Change the point coordinates in the model so that they appear in the simulated scene in ESIM.
  2. Change the camera trajectory to center it on your 3D model. For that, you may use the parameter `--trajectory_offset_x|y|z`.

Jee-King, 2020-07-04:
@yunxdai  方便加下qq/微信吗，有些模拟器的问题想咨询一下