uzh-rpg/rpg_esim #40: is it possible to simulate multiple images at the same time ?

wl082013, 2019-08-14:
Hi, I just tried the simulator  and it is nice.
but I am wondering whether it is possible to automatically read image folder (multiple images)? 
Changing the image path command in the config to read single image  is really cumbersome.
Thank you so much for your help.

supitalp, 2019-08-16:
Hello,
This is currently not possible. However, writing a script to automate the task is not too difficult.
Attached is an example script that shows how to do that. The script generates planar datasets similar to the ones in this paper: http://rpg.ifi.uzh.ch/docs/arXiv19_Rebecq.pdf .

Example script: [generate_synthetic_planar_dataset.zip](https://github.com/uzh-rpg/rpg_esim/files/3508463/generate_synthetic_planar_dataset.zip)

qwn19970728, 2019-10-15:
Hello, have you got the right script? Thank you!