uzh-rpg/rpg_esim #43: bug when contrast threshold < 0

eugenelyj, 2019-08-26:
It seems there will be a bug when contrast threshold smaller than zero (e.g `CHECK_GE(t, last_stamp_at_xy)` will failed). That is not rare. For example, if i set `constrast_threshold` to 0.10 and `contrast_sigma` to 0.05, it will be triggered often.

supitalp, 2019-09-09:
Thanks, this is indeed a bug. I will push a fix soon. Meanwhile, you can fix it by setting a minimum sampled contrast threshold by adding the following two lines after this line: https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim/src/event_simulator.cpp#L60, inside the if condition:

```
constexpr ImageFloatType minimum_contrast_threshold = 0.01;
C = std::max(minimum_contrast_threshold, C);
```

This should fix the issue. Do not forget to recompile before checking.
Thanks,
Henri

supitalp, 2019-09-16:
Fixed this: https://github.com/uzh-rpg/rpg_esim/commit/dc8a94f2b0d73aaf5a2b4af4d02759e027aa7d6b

closed 2019-09-16 by supitalp