uzh-rpg/rpg_esim #46: How can I make the camera trajectory?

jinho88, 2019-09-09:
I want to make the camera shaking to locate the object in the similar position.
I mean I want to move the camera fast and small amount. 

How can I make the camera trajectory?

supitalp, 2019-09-10:
Hello,
So you want to simulate a "shaking" camera motion.
You can achieve this by using the random trajectory generator based on splines.

- To achieve a "jerky" motion, reduce the spline order (`--trajectory_spline_order`) to get less smooth trajectories (a good value might be `2` in your case).
- To set the length of the trajectory, use the flag `--trajectory_length_s` (e.g. 2 seconds).
- To set the amplitude of the motion, use the trajectory multipliers (e.g. `--trajectory_multiplier_x/y=0.04`).
- To set the speed of the shaking motion, specify the number of segments in the trajectory with the parameter `--trajectory_num_spline_segments` (e.g. 50). This parameter controls the number of times that the camera will change direction, i.e. the higher the value, the faster the motion will be. 
-  Finally, set `--trajectory_lambda` to 0 (this parameter controls the maximum acceleration along the trajectory: 0 = no maximum acceleration and > 0 means the acceleration will be damped, resulting in smoother trajectories.

Here is an example config file that should yield a result similar to this: https://youtu.be/CDUWhd_Rl_k

Config file: `fast_shaking_example.conf`:

```yaml
--vmodule=data_provider_online_render=0,camera_simulator=0,planar_renderer=1,renderer_factory=1
--random_seed=0
--data_source=0
--path_to_output_bag=/tmp/out.bag

--contrast_threshold_pos=0.10
--contrast_threshold_neg=0.10
--contrast_threshold_sigma_pos=0.
--contrast_threshold_sigma_neg=0.
--refractory_period_ns=0

--exposure_time_ms=10.0
--use_log_image=1
--log_eps=0.01

--calib_filename=/home/user/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/calib/shaking_example.yaml
--renderer_hfov_cam_source_deg=90.0

--renderer_type=0
--renderer_texture=/home/user/henri/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_planar_renderer/textures/driving.png
--renderer_preprocess_gaussian_blur=0.25
--renderer_preprocess_median_blur=3
--renderer_plane_x=0.0
--renderer_plane_y=0.0
--renderer_plane_z=-0.8
--renderer_plane_qw=0.0
--renderer_plane_qx=1.0
--renderer_plane_qy=0.0
--renderer_plane_qz=0.0
--renderer_extend_border=0
--renderer_zmin=0.2

--trajectory_type=0
--trajectory_length_s=2.0
--trajectory_sampling_frequency_hz=100
--trajectory_spline_order=2
--trajectory_num_spline_segments=50
--trajectory_lambda=0.0
--trajectory_multiplier_x=0.04
--trajectory_multiplier_y=0.04
--trajectory_multiplier_z=0.
--trajectory_multiplier_wx=0.
--trajectory_multiplier_wy=0.
--trajectory_multiplier_wz=0.

--simulation_minimum_framerate=20.0
--simulation_imu_rate=1000.0
--simulation_adaptive_sampling_method=1
--simulation_adaptive_sampling_lambda=0.5

--ros_publisher_frame_rate=50
--ros_publisher_depth_rate=0
--ros_publisher_optic_flow_rate=0
--ros_publisher_pointcloud_rate=10
--ros_publisher_camera_info_rate=1
```

where the calibration file and `driving.png` files can be found in [this issue](https://github.com/uzh-rpg/rpg_esim/issues/45).

supitalp, 2019-09-16:
Dear @jinho88, did that help solve your issue?

jinho88, 2019-09-16:
Yes, It is very helpful. Thank you.