uzh-rpg/rpg_esim #49: Failed     << assimp_catkin:make                   [ Exited with code 2 ]

Lucksong, 2019-09-19:
There was a error "Failed     << assimp_catkin:make                   [ Exited with code 2 ]" when I run "catkin build esim_ros"

make[2]: *** [assimp_src-prefix/src/assimp_src-stamp/assimp_src-download] Error 1
make[1]: *** [CMakeFiles/assimp_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/sss/sim_ws/build/assimp_catkin; catkin build --get-env assimp_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << assimp_catkin:make                   [ Exited with code 2 ]                                                                                                                                  
Failed    <<< assimp_catkin                        [ 27 minutes and 6.6 seconds ]                                                                                                                          
Abandoned <<< imp_opengl_renderer                  [ Unrelated job failed ]                                                                                                                                
Abandoned <<< esim_data_provider                   [ Unrelated job failed ]                                                                                                                                
Abandoned <<< esim                                 [ Unrelated job failed ]                                                                                                                                
Abandoned <<< esim_ros                             [ Unrelated job failed ]                                                                                                                                
[build] Summary: 32 of 37 packages succeeded.                                                                                                                                                              
[build]   Ignored:   8 packages were skipped or are blacklisted.                                                                                                                                           
[build]   Warnings:  3 packages succeeded with warnings.                                                                                                                                                   
[build]   Abandoned: 4 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 27 minutes and 13.5 seconds total.                                                                                                                                                        
[build] Note: Workspace packages have changed, please re-source setup files to use them.

supitalp, 2019-09-19:
Hello,
It looks like the source code of the dependency `assimp` could not be downloaded. Do you have a stable Internet connection?
Please see [this issue as well](https://github.com/uzh-rpg/rpg_esim/issues/42).

Lucksong, 2019-09-19:
Thanks for your replay
I have seened the issue, and I download the assimp manually then I went to the path "build/assimp_catkin" and run the code "catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j", but it costed so much time. It has been four hours but is still running and not any logs. Is it normal？

supitalp, 2019-09-19:
Four hours to compile  assimp seems like an abnormally long time. It should take a couple of minutes, not hours. I am not sure however what could go wrong there.
Could you try remove the `assimp_catkin` build folder:

```rm -rf ~/sim_ws/build/assimp_catkin```

and try compiling assimp again:

```catkin build assimp_catkin```

It it succeeds, then you can continue building ESIM:

```catkin build esim_ros```

Lucksong, 2019-09-19:
Ok, thanks  ,I will try it.

Lucksong, 2019-09-19:
Thanks very much! It was solved and now I am trying your test example.By the way , didn`t you provide the script generate_stamps_file.py?

supitalp, 2019-09-19:
Great to hear!
Yes, the script is available there: https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/scripts/generate_stamps_file.py

If you do not have it in your local install, please pull (I uploaded it recently) and recompile ESIM (`catkin build esim_ros`, do not worry if will need to recompile a few things, not the entire project).

Lucksong, 2019-09-19:
Thanks very much

Lucksong, 2019-09-19:
[rospack] Error: package 'dvs_renderer' not found
In the test example, the error occurred when I run "rosrun dvs_renderer dvs_renderer events:=/cam0/events"

supitalp, 2019-09-19:
Looks like the `dvs_renderer` node was not built. Run: `catkin build
dvs_renderer`, make sure to source your workspace again (`source
~/sim_ws/devel/setup.bash`), and run again.

Lucksong, 2019-09-19:
Did the code need some time? I run it with no error but not any result now

Lucksong, 2019-09-19:
I run the code "rosrun dvs_renderer dvs_renderer events:=/cam0/events" and no result now

supitalp, 2019-09-19:
Please have a look at my answer here: https://github.com/uzh-rpg/rpg_esim/issues/48#issuecomment-533009294 , it should help

yujeong-star, 2020-04-01:
Hi I have a similar error. In my case glog_catkin is the one making the error.
I removed the glog_catkin folder and re built it but that makes the error too.

Full error description is as following:
1. Error after run "catkin build esim_ros"
_______________________________________________________________________________
Errors     << glog_catkin:make /home/yujeong/sim_ws/logs/glog_catkin/build.make.000.log
/bin/sh: 1: autoreconf: not found
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-configure] Error 127
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/yujeong/sim_ws/build/glog_catkin; catkin build --get-env glog_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << glog_catkin:make                     [ Exited with code 2 ]      
Failed    <<< glog_catkin                          [ 3.1 seconds ]             
Abandoned <<< eigen_checks                         [ Unrelated job failed ]    
Abandoned <<< esim_unrealcv_bridge                 [ Unrelated job failed ]    
Abandoned <<< minkindr                             [ Unrelated job failed ]    
Abandoned <<< minkindr_conversions                 [ Unrelated job failed ]    
Abandoned <<< ze_common                            [ Unrelated job failed ]    
Abandoned <<< imp_core                             [ Unrelated job failed ]    
Abandoned <<< imp_bridge_opencv                    [ Unrelated job failed ]    
Abandoned <<< imp_bridge_ros                       [ Unrelated job failed ]    
Abandoned <<< ze_cameras                           [ Unrelated job failed ]    
Abandoned <<< esim_common                          [ Unrelated job failed ]    
Abandoned <<< esim_rendering                       [ Unrelated job failed ]    
Abandoned <<< esim_visualization                   [ Unrelated job failed ]    
Abandoned <<< imp_multi_objects_2d                 [ Unrelated job failed ]    
Abandoned <<< imp_opengl_renderer                  [ Unrelated job failed ]    
Abandoned <<< imp_panorama_renderer                [ Unrelated job failed ]    
Abandoned <<< imp_planar_renderer                  [ Unrelated job failed ]    
Abandoned <<< imp_unrealcv_renderer                [ Unrelated job failed ]    
Abandoned <<< ze_matplotlib                        [ Unrelated job failed ]    
Abandoned <<< ze_ros                               [ Unrelated job failed ]    
Abandoned <<< ze_visualization                     [ Unrelated job failed ]    
Abandoned <<< ze_splines                           [ Unrelated job failed ]    
Abandoned <<< ze_vi_simulation                     [ Unrelated job failed ]    
Abandoned <<< esim_trajectory                      [ Unrelated job failed ]    
Abandoned <<< esim_data_provider                   [ Unrelated job failed ]    
Abandoned <<< esim                                 [ Unrelated job failed ]    
Abandoned <<< esim_ros                             [ Unrelated job failed ] 
[build] Summary: 10 of 37 packages succeeded.                                  
[build]   Ignored:   8 packages were skipped or are blacklisted.               
[build]   Warnings:  4 packages succeeded with warnings.                       
[build]   Abandoned: 26 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4 minutes and 23.8 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.

2. Error after remove glog_catkin folder and run "catkin build glog_catkin"
_______________________________________________________________________________
Errors     << glog_catkin:make /home/yujeong/sim_ws/logs/glog_catkin/build.make.001.log
/bin/sh: 1: autoreconf: not found
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-configure] Error 127
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/yujeong/sim_ws/build/glog_catkin; catkin build --get-env glog_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << glog_catkin:make             [ Exited with code 2 ]              
Failed    <<< glog_catkin                  [ 2.2 seconds ]                     
[build] Summary: 2 of 3 packages succeeded.                                    
[build]   Ignored:   41 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.4 seconds total. 

Thanks in advance!

erlini, 2020-07-05:
hello!
I cannot compile the assimp library even though I have tried all the above suggestions. I found my network capable of downloading the "assimp-4.0.1.zip", but I still can not run "catkin build assimp_catkin" successfuly. I have no idea about how to fix it.

erlini, 2020-07-05:
@supitalp

erlini, 2020-07-31:
Already addressed.

reddy528, 2020-09-01:
> Already addressed.

how did you solve it?