uzh-rpg/rpg_esim #50: ERROR: cannot launch node of type [esim_ros/esim_node]: can't locate node [esim_node] in package [esim_ros]

himanshu9345, 2019-09-19:
Hello, All
I have completed installation of rpg_esim (https://github.com/uzh-rpg/rpg_esim/wiki/Installation)
I am trying to install Planar Renderer (https://github.com/uzh-rpg/rpg_esim/wiki/Planar-Renderer).

1. Run two commands as follows
```
~/sim_ws/src/rpg_esim/event_camera_simulator$ roscd esim_ros/
himanshu@himanshu-pc:~/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros$ gedit cfg/example.conf
```
2. Then i changed the absolute path
[image]

3. Run below code
`roslaunch esim_ros esim.launch config:=cfg/example.conf`

But, I got below error.
[image]

supitalp, 2019-09-19:
Hello,
Have you sourced your ROS workspace after compiling?
Please try do it as follows: `source ~/sim_ws/devel/setup.bash` and try to run ESIM again.

himanshu9345, 2019-09-19:
Yes, I did it. 
I even used provided cmd but still same error.

supitalp, 2019-09-19:
Hello,
OK. Did the installation process succeed (without yielding errors)? If the ROS workspace is sourced, and that esim_node cannot be found, it may be because it was not compiled.

supitalp, 2019-09-19:
Also, what happens when you run this?

`rosrun esim_ros esim_node`

himanshu9345, 2019-09-19:
Thank you, @supitalp. I recompiled it, and it worked. There was a compilation error before which I missed.

closed 2019-09-19 by himanshu9345