uzh-rpg/rpg_esim #51: How to get the trajectory's moving function?

qwn19970728, 2019-09-27:


supitalp, 2019-09-27:
Hello,
The camera trajectory is published on the `/tf` topic, and saved in the `/cam0/pose` topic in the saved rosbag. Does this answer your question?

qwn19970728, 2019-09-29:
> Hello,
> The camera trajectory is published on the `/tf` topic, and saved in the `/cam0/pose` topic in the saved rosbag. Does this answer your question?

> Hello,
> The camera trajectory is published on the `/tf` topic, and saved in the `/cam0/pose` topic in the saved rosbag. Does this answer your question?

> Hello,
> The camera trajectory is published on the `/tf` topic, and saved in the `/cam0/pose` topic in the saved rosbag. Does this answer your question?

Thank you!

qwn19970728, 2019-09-29:
> Hello,
> The camera trajectory is published on the `/tf` topic, and saved in the `/cam0/pose` topic in the saved rosbag. Does this answer your question?

Thank you!

closed 2019-09-30 by supitalp