uzh-rpg/rpg_esim #52: how to enable multi-camera systems?

LavieC, 2019-10-10:
Hi,

“Support for multi-camera systems” is mentioned in the README file, and I have noticed that there are some lines regarding camera numbers in the code. Yet I cannot figure out the exact way to enable it. 

So my problem is that is this multi-camera system means a simulation of two physical cameras in different spatial positions, and how can to achieve this simulation.

Thks!

supitalp, 2019-10-10:
Hello,
ESIM can simulate a system composed of an arbitrary number of cameras rigidly attached to a rig (for example, a stereo rig). The cameras can of course be at different spatial positions with respect to the rig, though they move "synchronously" with the rig.

The geometry of the rig must be defined in the camera calibration file, with one `-camera` item per camera in the rig. The relative position/orientation of the camera with respect to the center of the rig is defined through the `T_B_C` matrix (4x4 transformation matrix from Camera frame to Body (rig) frame).

For example, to simulate a stereo camera setup with two parallel cameras separated by 15cm, the calibration file might look like:

`pinhole_stereo.yaml`
```yaml
label: "stereo_rig"
id: 0
cameras:
- camera:
    label: dvs0
    id: 1
    line-delay-nanoseconds: 0
    image_height: 180
    image_width: 240
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [200.0, 200.0, 120.0, 90.0]
    distortion:
      type: none
      parameters:
        cols: 1
        rows: 0
        data: []
  T_B_C:
    cols: 4
    rows: 4
    data: [1, 0, 0,  -0.075,
           0, -1, 0, 0,
           0, 0, -1, 0,
           0, 0, 0, 1]
- camera:
    label: dvs1
    id: 2
    line-delay-nanoseconds: 0
    image_height: 180
    image_width: 240
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [200.0, 200.0, 120.0, 90.0]
    distortion:
      type: none
      parameters:
        cols: 1
        rows: 0
        data: []
  T_B_C:
    cols: 4
    rows: 4
    data: [1, 0, 0,  0.075,
           0, -1, 0, 0,
           0, 0, -1, 0,
           0, 0, 0, 1]
``` 

Do not forget to update the path to the calibration file in your config file (`--calib_filename`).
Upon launching, ESIM will automatically spawn the required number of cameras, and publish events and images for each camera on the `/cam0`, `/cam1`, etc. namespaces.

Hope that helps, let me know if you have any questions.

LavieC, 2019-11-06:
Thank a lot!