uzh-rpg/rpg_esim #56: Check failed: images_in_str_.is_open()

andrewcfreeman, 2019-11-12:
When trying to simulate events from a video frame folder, I get the following error:

F1112 09:58:52.770164 29563 data_provider_from_folder.cpp:15] Check failed: images_in_str_.is_open() 
*** Check failure stack trace: ***
    @     0x7ff669deab0d  google::LogMessage::Fail()
    @     0x7ff669dec9b1  google::LogMessage::SendToLog()
    @     0x7ff669dea63d  google::LogMessage::Flush()
    @     0x7ff669ded369  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ff66a8e5900  event_camera_simulator::DataProviderFromFolder::DataProviderFromFolder()
    @     0x7ff66a8cb3b8  event_camera_simulator::loadDataProviderFromGflags()
    @     0x5590e2bec253  (unknown)
    @     0x7ff669248b97  __libc_start_main
    @     0x5590e2becdba  (unknown)
Aborted (core dumped)

I've verified that images.csv is present and the permissions are correct. Everything was working fine before, so I'm not sure what changed.

realWDC, 2020-02-06:
I am suffering from the same issue... Did you manage to solve this? Thanks!

andrewcfreeman, 2020-02-11:
No, unfortunately. Still have no clue.

ihollywhy, 2020-02-27:
Maybe you provide a wrong path to the input?

aigooman, 2020-03-20:
I first use "~/path/to/data" and got this error. 
Then I changed it to "/home/user/path/to/data" and ran successfully.
Hope this helps.