uzh-rpg/rpg_esim #60: CED color event format

smmmmi, 2019-12-25:
Hi @supitalp sorry to post here. 
I am following the [CED](http://rpg.ifi.uzh.ch/CED.html) dataset.
What format does the color event data follow?
How can  RGBG events be extracted from the stream of events (eg indoors_office.bag).
It looks like it contains normal events (/dvs/events) and color images (/dvs/image_color) only but not the color events. If (/dvs/events) contains the info how is it coded? Is it the same for ESIM?
Thanks.

supitalp, 2019-12-26:
Hi @smmmmi,
The event data in the CED dataset is exactly the same as for regular, monochrome event data. The only difference between the DAVIS346color sensor and a monochrome sensor is that there is a color filter array (Bayer pattern) in front of the sensor, which means that some pixels will be more sensitive to a certain color (red, blue or green, depending on the pixel position in the array).
An example of how to "filter" events according to their sensitive color can be found here: https://github.com/uzh-rpg/rpg_e2vid/blob/master/image_reconstructor.py#L72 (lines 72 to 75).

ESIM follows the exact same convention as the DAVIS346color, so no changes should be required.
Note that I never tried to generate color data from ESIM and train a network for color event processing (e.g. image reconstruction) directly on the synthetic color events, so I'm not sure how well this would work.

Hope that helps,
Henri

smmmmi, 2020-01-20:
Great, thanks for the reply.

closed 2020-01-20 by smmmmi