uzh-rpg/rpg_esim #61: Feature/color branch won't install

realArtichoke, 2020-01-29:
Hi, I am trying to install feature/color branch and I have trouble with building.
```
Errors     << esim_ros:make /home/imagingtre/catkin_ws/logs/esim_ros/build.make.014.log
/home/imagingtre/catkin_ws/build/esim_ros/setup_custom_pythonpath.sh: 5: exec: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: Permission denied
make[2]: *** [/home/imagingtre/catkin_ws/devel/.private/esim_ros/include/esim_ros/example_colorConfig.h] Error 126
make[1]: *** [CMakeFiles/esim_ros_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/imagingtre/catkin_ws/build/esim_ros; catkin build --get-env esim_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << esim_ros:make                        [ Exited with code 2 ]      
Failed    <<< esim_ros                             [ 0.6 seconds ]             
[build] Summary: 35 of 36 packages succeeded.                                  
[build]   Ignored:   21 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 11.2 seconds total.                         
```

In similar issues I've seen an advice to execute  chmod +x for that cfg file. But then I get this whilst building:

```
Errors     << esim_ros:make /home/imagingtre/catkin_ws/logs/esim_ros/build.make.015.log
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 1: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --vmodule=data_provider_online_render=0,camera_simulator=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 2: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --random_seed=50: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 3: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --data_source=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 4: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --path_to_output_bag=/tmp/out.bag: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 6: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_pos=0.15: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 7: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_neg=0.15: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 8: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_sigma_pos=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 9: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --contrast_threshold_sigma_neg=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 11: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --exposure_time_ms=20.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 12: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --use_log_image=true: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 13: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --log_eps=0.001: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 14: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --simulate_color_events=true: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 16: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --calib_filename=/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/calib/pinhole_mono_nodistort.yaml: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 18: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_type=0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 19: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_texture=/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/imp/imp_planar_renderer/textures/carpet.jpg: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 20: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_hfov_cam_source_deg=85.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 21: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_preprocess_gaussian_blur=0.25: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 22: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_preprocess_median_blur=11: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 23: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_x=0.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 24: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_y=0.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 25: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_z=-1.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 26: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_qw=0.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 27: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_qx=1.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 28: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_qy=0.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 29: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_plane_qz=0.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 30: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_extend_border=1: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 31: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --renderer_zmin=1.0: not found
/home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: 33: /home/imagingtre/catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/example_color.cfg: --trajectory_type=0: not found
[... cut ...]
[build] Summary: 35 of 36 packages succeeded.                                 
[build]   Ignored:   21 packages were skipped or are blacklisted.             
[build]   Warnings:  None.                                                    
[build]   Abandoned: None.                                                    
[build]   Failed:    1 packages failed.                                       
[build] Runtime: 10.2 seconds total.          
```
I am using ROS Melodic if it's important.
I will be very grateful for any help with that.

DenisDiachkov, 2020-02-07:
The problem has been solved by moving example_color.cfg from its initial folder to somewhere else.
It works but I am not sure if everything is fine and nothing depends on it.

cxfnyang, 2020-05-20:
> The problem has been solved by moving example_color.cfg from its initial folder to somewhere else.
> It works but I am not sure if everything is fine and nothing depends on it.

I meet the same problem, did you find any solution? I will be very grateful if you can share it

DenisDiachkov, 2020-05-20:
@cxfnyang 
I didn't find proper solution for that but after moving the file out of the folder everything worked and I didn't meet any problems with it further.

Climmy, 2022-07-11:
Renaming "example_color.cfg" to "example_color.conf" could solve the problem.