uzh-rpg/rpg_esim #62: Render settings

realArtichoke, 2020-02-03:
Hi,
I've got a problem with [UE4 render](https://github.com/uzh-rpg/rpg_esim/wiki/Photorealistic-Rendering-Engine-based-on-Unreal-Engine). I get some roughness when I try to run 

> roslaunch esim_ros esim.launch config:=cfg/unrealcv.conf

and extract images from the .bag file.
[image]

I want to make it looking smooth and more photorealistic. I tried to change [this file](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_unrealcv_bridge/src/unrealcv_bridge.cpp) with adding more of anti aliasing and played with many other parameters. But it doesn't make any difference.

It seems needed parameters are overrided somewhere below this code, probably in some config files. But I can't find any render options in config files related to unrealcv.

What should I do to remove these rough edges? What file should I play with?

DenisDiachkov, 2020-02-04:
The solution: to manipulate the render settings with  [UE4 parameters](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_unrealcv_bridge/src/unrealcv_bridge.cpp#L54) you need to set  ``` --vmodule=unrealcv_renderer=1 ``` instead of 0 in [unrealcv.conf](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_ros/cfg/unrealcv.conf)

DenisDiachkov, 2020-02-07:
I was wrong. Previous change doesn't make difference for images from the .bag file but only changes image quality in playground.sh window. The rest is the same.

DenisDiachkov, 2020-03-06:
The problem is solved.

I changed width and height to 4k resolution In RealisticRender/Binaries/unrealcv.ini file.

closed 2020-03-06 by DenisDiachkov