uzh-rpg/rpg_esim #63: Different cameras focal lengths

realArtichoke, 2020-02-04:
Hi,
I've got a question during testing [that thing](https://github.com/uzh-rpg/rpg_esim/wiki/Photorealistic-Rendering-Engine-based-on-Unreal-Engine).

In playground.sh window I see that picture 
[image]
But when I look at rqt visualisation or save frames from the .bag file I see this
[image]
[image]

Two different images are going to the screen and the .bag file. It seems they have different cameras  intrinsic parameters. How to fix it and get the same images?

Thank you.

supitalp, 2020-02-04:
Hi,
I am not able to reproduce that behavior. Could you share the calibration file used? The aspect ratio of the images looks weird.
Thanks,
Henri

DenisDiachkov, 2020-02-05:
Sure!

pinhole_mono_nodistort_forward.yaml
```
label: "simulated_camera"
id: 433585ebda8d1431223927a14788a127
cameras:
- camera:
    label: dvs0
    id: a0fba5412e961934d842d3a2a78e5cba
    line-delay-nanoseconds: 0
    image_height: 540
    image_width: 960
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [480.0, 480.0, 480, 270.0]
    distortion:
      type: none
      parameters:
        cols: 1
        rows: 0
        data: []
  T_B_C:
    cols: 4
    rows: 4
    data: [-1, 0, 0,  0,
           0, 0, -1, 0,
           0, -1, 0, 0,
           0, 0, 0, 1]
```
unrealcv.conf
```
--data_source=0
--vmodule=unrealcv_renderer=1
--path_to_output_bag=/home/imagingtre/grayscale_catkin_ws/out.bag
--random_seed=1

--contrast_threshold_pos=0.6
--contrast_threshold_neg=0.6
--contrast_threshold_sigma_pos=0
--contrast_threshold_sigma_neg=0

--exposure_time_ms=5.0
--use_log_image=1
--log_eps=0.001

--calib_filename=/home/imagingtre/grayscale_catkin_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/calib/pinhole_mono_nodistort_forward.yaml

--renderer_type=3

--trajectory_type=0
--trajectory_length_s=100.0
--trajectory_sampling_frequency_hz=5
--trajectory_spline_order=5
--trajectory_num_spline_segments=100
--trajectory_lambda=0.1
--trajectory_multiplier_x=1.0
--trajectory_multiplier_y=1.0
--trajectory_multiplier_z=0.5
--trajectory_multiplier_wx=0.15
--trajectory_multiplier_wy=0.15
--trajectory_multiplier_wz=1.

--x_offset=0.0
--y_offset=0.0
--z_offset=1.0

--simulation_minimum_framerate=5.0
--simulation_imu_rate=1000.0
--simulation_adaptive_sampling_method=1
--simulation_adaptive_sampling_lambda=2.0

--ros_publisher_frame_rate=300
--ros_publisher_depth_rate=300
--ros_publisher_optic_flow_rate=200
--ros_publisher_pointcloud_rate=50
--ros_publisher_camera_info_rate=10
```

DenisDiachkov, 2020-02-07:
Update: I tried to capture the image what unrealCV generates before translating it into grayscale [here](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/imp/imp_unrealcv_renderer/src/unrealcv_renderer.cpp#L122). I got this
[image]
It is 2000x2000 pixels image. And its FOV is more than playnground.sh window shows. It seems playnground.sh cuts the image a bit for some reason.

Below this code the image is resized to the camera settings(from the config file I suppose). That's why  
the proportion is getting different. I tried to assign SetRes UE4 parameter [here](https://github.com/uzh-rpg/rpg_esim/blob/master/event_camera_simulator/esim_unrealcv_bridge/src/unrealcv_bridge.cpp#L54) with sendCommand function. But the image from unrealCV is stil 2000x2000. 

How can I set it to another resolution?