uzh-rpg/rpg_esim #65: Start with the Planar Rendering Engine: “[esim_node-1] process has died”

tpzou, 2020-02-12:
Well, when I try to follow the step : 
            ` roslaunch esim_ros esim.launch config:=cfg/example.conf`
I get the following output:
              `process[esim_node-1]: started with pid [32677]
process[dvs_renderer-2]: started with pid [32678]
process[optic_flow_viz-3]: started with pid [32679]
F0212 20:36:24.050110 32677 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.
*** Check failure stack trace: ***
    @     0x7ff1669f743d  google::LogMessage::Fail()
    @     0x7ff1669f9253  google::LogMessage::SendToLog()
    @     0x7ff1669f6fcb  google::LogMessage::Flush()
    @     0x7ff1669f9c3e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ff1621d27b7  ze::cameraRigFromGflags()
    @     0x7ff167fd7a2f  event_camera_simulator::DataProviderOnlineMoving3DCameraRig::DataProviderOnlineMoving3DCameraRig()
    @     0x7ff167fd0087  event_camera_simulator::loadDataProviderFromGflags()
    @           0x404dfe  main
    @     0x7ff165e87830  __libc_start_main
    @           0x405779  _start
[esim_node-1] process has died [pid 32677, exit code -6, cmd /home/ztp/sim_ws/devel/lib/esim_ros/esim_node --v=1 --vmodule=data_provider_from_folder=10 --flagfile=/home/ztp/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/opengl.conf __name:=esim_node __log:=/home/ztp/.ros/log/539ea9b8-4d90-11ea-bd58-2016d82c5921/esim_node-1.log].
log file: /home/ztp/.ros/log/539ea9b8-4d90-11ea-bd58-2016d82c5921/esim_node-1*.log
[INFO] [1581510984.571403]: Started flow converter node
[INFO] [1581510984.571867]: Step size between arrows: 7
[INFO] [1581510984.572202]: Scale factor: 0.07
[INFO] [1581510984.572485]: Upsample factor: x1
[INFO] [1581510984.572758]: Publish rate: 100 Hz
`

closed 2020-02-12 by tpzou

mohammed-amr, 2020-02-26:
I see you've closed the issue. You manage to figure out what was wrong?

YuxuanSnow, 2021-04-06:
Could you please mention how you solve the issue? I am facing the issue now. Thanks.