uzh-rpg/rpg_esim #73: Trajectories

DominikVincent, 2020-04-24:
Hello,

maybe I just do not have enough foreknowledge, but I am confused about trajectories and could not find sufficient documentation. If you could point me to a some sources  or answer a few questions that would help.

1. the blender exported .obj will not influence the camera trajectory at all, correct?
2. The blender exported .obj will be static, independent of animations within the blender scene, correct?
3. There are two trajectory types random and spline interpolation between user defined points in the csv file is --trajectory_type=0 random and 1 csv? 
by checking the code i fgured to add trajectory_csv_file to the conf file, is that correct? And is there really no documentation for it or did i miss it?
4. When running the example opengl with the quadrocopter or poster csv, the optical flow map is not aligned with the video, when inspecting with rqt. Is that an bug or something i missed?
5. How can you easly generate the csv files for trajectories?

Thanks a lot for any feedback for my many questions.

Jee-King, 2020-07-04:
@DominikVincent  Hi, can you use _OpenGL Rendering Engine_ smoothly? Could you point me to a complete process. Without detailed instructions, it is difficult for me to use it.

milmario, 2020-11-03:
@DominikVincent i suggest to create the path of the camera directly in the scene in blender. Then, you can export the location of the camera at each frame to a .csv file. I wrote my own python code, which i inserted in the scripting-part of blender:

```

import bpy
import numpy as np

def get_location_rotation_from_path(scene, camera, fps):
    arr = np.empty((0,8), float)
    ts = 0
    #myRange = range(scene.frame_start, scene.frame_end)
    myRange = range(0,20)
    for fr in myRange:
        scene.frame_set(fr)

        q_norm = np.linalg.norm(cam.matrix_world.decompose()[1][:])

        x = cam.matrix_world.decompose()[0][0]
        y = cam.matrix_world.decompose()[0][1]
        z = cam.matrix_world.decompose()[0][2]
        qx = abs(cam.matrix_world.decompose()[1][1]/q_norm)
        qy = abs(cam.matrix_world.decompose()[1][2]/q_norm)
        qz = abs(cam.matrix_world.decompose()[1][3]/q_norm)
        qw = abs(cam.matrix_world.decompose()[1][0]/q_norm)
        if cam.matrix_world.decompose()[1][0]/q_norm < 0 or cam.matrix_world.decompose()[1][1]/q_norm <0 or cam.matrix_world.decompose()[1][2]/q_norm<0 or cam.matrix_world.decompose()[1][3]/q_norm:
            print("Trajectory at frame " + str(fr) + " is smaller than 0!")
        tmp_arr = np.hstack((ts, x,y,z,qx,qy,qz,qw))

        arr = np.vstack((arr, tmp_arr)) 
        ts += 1/fps*1000000000
    return arr
    

cam = bpy.data.objects["Camera"]
scene = bpy.data.scenes["Scene"]
fps = 20

arr = get_location_rotation_from_path(scene, cam, fps)

arr = np.around(arr, decimals = 6)    
print(arr)
header = "timestamp, x, y, z, qx, qy, qz, qw"
np.savetxt("/path_to_output_folder/camera_trajectory.csv", arr, delimiter=', ', header=header,fmt='%1.6f')
print('all done!')
```

Note the following:

- I'm having some issues with the quaternions, when they cross the 0 coordinate... I don't have a fix for this yet (this is why i used the absolute value above)
- The quaternion norm always has to be 1 (included in my code snippet)

I hope I could help you at least a bit.

DominikVincent, 2020-11-07:
Thanks for the answer. However, I am no longer working on my project, but hopefully, it will be useful for others.

celynw, 2021-01-11:
Thanks @milmario, really helpful. Works fine for me once I sorted out some blender `numpy` issues.
At the end of the line with the big conditional, I think you missed a bit on the last condition, and it should end like:
```
cam.matrix_world.decompose()[1][3]/q_norm<0:
```