uzh-rpg/rpg_esim #75: F0530 17:50:01.090956 16067 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.

SeaEastXu, 2020-05-30:
Hi!
I have a similar issue to https://github.com/uzh-rpg/rpg_esim/issues/10.
I have followed @supitalp 's opinion. https://github.com/uzh-rpg/rpg_esim/issues/10#issuecomment-446555859
But I haven't modified the file,it's default.

Here is my error output:
$ rosrun esim_ros esim_node \ --data_source=2 \ --path_to_output_bag=~/sim_ws/tmp/out.bag \ --path_to_data_folder=/home/seaeast/sim_ws/tmp/demo/frames/ \ --ros_publisher_frame_rate=10 \ --exposure_time_ms=10.0 \ --use_log_image=1 --log_eps=0.1 --contrast_threshold_pos=0.15 --contrast_threshold_neg=0.15
F0530 16:07:07.848632 8489 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.
*** Check failure stack trace: ***
@ 0x7fe5b4757b0d google::LogMessage::Fail()
@ 0x7fe5b47599b1 google::LogMessage::SendToLog()
@ 0x7fe5b475763d google::LogMessage::Flush()
@ 0x7fe5b475a389 google::LogMessageFatal::~LogMessageFatal()
@ 0x7fe5b220e817 ze::cameraRigFromGflags()
@ 0x7fe5b52458b0 event_camera_simulator::DataProviderOnlineMoving3DCameraRig::DataProviderOnlineMoving3DCameraRig()
@ 0x7fe5b5238457 event_camera_simulator::loadDataProviderFromGflags()
@ 0x55af8aafe223 (unknown)
@ 0x7fe5b3bb5b97 __libc_start_main
@ 0x55af8aafed8a (unknown)
Aborted

Any help would be really appreciated
thanks

SeaEastXu, 2020-05-30:
There are no problems in `catkin build esim_ros` progress. And my ROS version is melodic.

There is another error output:

seaeast@seaeast-XPS-15-9560:~/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_multi_objects_2d$ roslaunch esim_ros esim.launch  config:=cfg/multi_objects.conf
... logging to /home/seaeast/.ros/log/63c9eaaa-a245-11ea-b4dc-9cb6d0e46b2b/roslaunch-seaeast-XPS-15-9560-16621.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://seaeast-XPS-15-9560:32839/

SUMMARY
========

PARAMETERS
 * /optic_flow_viz/arrows_scale: 0.07
 * /optic_flow_viz/arrows_step: 7
 * /optic_flow_viz/arrows_upsample_factor: 1
 * /optic_flow_viz/publish_rate: 100
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

ROS_MASTER_URI=http://localhost:11311

process[esim_node-1]: started with pid [16637]
process[dvs_renderer-2]: started with pid [16638]
process[optic_flow_viz-3]: started with pid [16639]
F0530 17:54:08.533686 16637 file_utils.hpp:43] Check failed: fileExists(filename) File does not exist: /home/user/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_multi_objects_2d/scenes/example.scene
*** Check failure stack trace: ***
    @     0x7f5ac1c64b0d  google::LogMessage::Fail()
    @     0x7f5ac1c669b1  google::LogMessage::SendToLog()
    @     0x7f5ac1c6463d  google::LogMessage::Flush()
    @     0x7f5ac1c67389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f5abfb5ecb7  event_camera_simulator::MultiObject2DRenderer::MultiObject2DRenderer()
    @     0x7f5ac276ccf9  event_camera_simulator::loadSimpleRendererFromGflags()
    @     0x7f5ac275cc97  event_camera_simulator::DataProviderOnlineSimple::DataProviderOnlineSimple()
    @     0x7f5ac27454b4  event_camera_simulator::loadDataProviderFromGflags()
    @     0x555b16860223  (unknown)
    @     0x7f5ac10c2b97  __libc_start_main
    @     0x555b16860d8a  (unknown)
[esim_node-1] process has died [pid 16637, exit code -6, cmd /home/seaeast/sim_ws/devel/lib/esim_ros/esim_node --v=1 --vmodule=data_provider_from_folder=10 --flagfile=/home/seaeast/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/cfg/multi_objects.conf __name:=esim_node __log:=/home/seaeast/.ros/log/63c9eaaa-a245-11ea-b4dc-9cb6d0e46b2b/esim_node-1.log].
log file: /home/seaeast/.ros/log/63c9eaaa-a245-11ea-b4dc-9cb6d0e46b2b/esim_node-1*.log
[INFO] [1590832449.016783]: Started flow converter node
[INFO] [1590832449.019615]: Step size between arrows: 7
[INFO] [1590832449.022005]: Scale factor: 0.07
[INFO] [1590832449.024274]: Upsample factor: x1
[INFO] [1590832449.026559]: Publish rate: 100 Hz

Aku02, 2020-07-10:
Hi!
I'm also facing the same issue any leads for resolving the same
Thanks

LarryDong, 2021-02-01:
You should change path to the camera calib file:
[https://github.com/uzh-rpg/rpg_esim/blob/d7708fff4acffec1771297674c4b917b7683d16b/event_camera_simulator/esim_ros/cfg/unrealcv.conf#L15](https://github.com/uzh-rpg/rpg_esim/blob/d7708fff4acffec1771297674c4b917b7683d16b/event_camera_simulator/esim_ros/cfg/unrealcv.conf#L15)

YuxuanSnow, 2021-04-06:
As LarryDong said, you should change the absolute path in .conf file. More specifically, change "user" to your computer name.

YuxuanSnow, 2022-08-05:
> 
Well, that's depend on where do you install and where is your ROS workspace.

You have to find this file in your computer, and change it according to it (`pwd`).

VinoDeng, 2022-08-06:
Thanks a lot, I have completed event camera simulation successfully via your method