uzh-rpg/rpg_esim #79: Ubuntu 20.04 (WSL), ROS Noetic - Simulating events from a video

zenmar, 2020-11-15:
Hi, I tried installing ESIM on WSL (Ubuntu 20.04) with ROS Noetic.

I then followed the [tutorial to simulate events from a video](https://github.com/uzh-rpg/rpg_esim/wiki/Simulating-events-from-a-video). 

Everything seems to work, up to the visualization part:

> You can now visualize the simulated events by opening rqt_image_view (in another terminal): rqt_image_view /dvs_rendering.

I don't have this /dvs_rendering folder/file, is that something I would need to implement myself, did something change since then, or might it even be a compatibility issue ?

Thank you!

closed 2020-11-19 by zenmar

sophiaas, 2020-11-30:
@zenmar Did you get the simulator working with ROS Noetic?

liufinback, 2023-07-13:
You need open another window and run the command below
rosrun dvs_renderer dvs_renderer events:=/cam0/events