uzh-rpg/rpg_esim #80: Support for ROS Noetic and Python3?

sophiaas, 2020-11-25:
Are there plans to make the event simulator compatible with ROS Noetic and Python 3? Has anyone tried that setup?

amenon4, 2020-12-15:
Yep, it does indeed work.

Dajamante, 2021-03-17:
Hello, since this issue is still open, did anyone manage to make it work?
We were forced to skip dependencies: pcl_ros, glfw, glm and ros-kinetic-hector-trajectory-server. Of course when running "ssim" nothing happens.

Our configurations:
Mac OS Catalina and Mac OS Big Sur

berndpfrommer, 2021-04-06:
I was able to get it to build on Ubuntu 20.04 ROS noetic, by tracking down the dependencies one-by-one. The pangolin_catkin needed modification in CMakeLists.txt:
```
  GIT_TAG origin/master
#  GIT_TAG 8b6a6b706d50d4172eeb6d5827f40b2207483cf0
```
And I needed to install python2.7-dev.
Well, it compiles, now I'll find out if it actually works....

Youras, 2021-05-10:
For ROS **Noetic** the building fails at **ze_splines**, at least for me.

The solution to this is described here  https://github.com/uzh-rpg/rpg_esim/issues/7#issuecomment-687999099.
In my case the building succeeded after the following commands:
```
mkdir ~/anaconda3/libuuid
mv ~/anaconda3/lib/libuuid* ~/anaconda3/libuuid
catkin build  esim_ros
```

Simulating events from a video (https://github.com/uzh-rpg/rpg_esim/wiki/Simulating-events-from-a-video) works well for me.

armzimmer, 2022-09-21:
@PingHGao I had the same issue, but was able to build, after editing the CMakeLists.txt file at rpg_esim/event_camera_simulator/esim_visualization/CMakeLists.txt by adding the line:
set(CMAKE_CXX_STANDARD_REQUIRED ON)

Found the solution [here](https://github.com/PointCloudLibrary/pcl/issues/3104#issuecomment-495889119)

gaopinghai, 2024-03-08:
> set(CMAKE_CXX_STANDARD_REQUIRED ON)

Thanks, it works！