uzh-rpg/rpg_esim #86: Not able to rectify this error: F0212 12:20:05.095577  5250 camera_rig.cpp:106] Check failed:  "fileExists(FLAGS_calib_filename) Camera file does not exist. "

Rishabh-Samra, 2021-02-12:
I wanted to use ESim simulator to get events from a video. ([https://github.com/uzh-rpg/rpg_esim/wiki/Simulating-events-from-a-video](url)) But I am stuck in the mid and getting the error: "**F0212 12:20:05.095577  5250 camera_rig.cpp:106] Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist.**" 
I have followed @supitalp 's suggestion and have changed the address of the calib file in unrealcv.conf according to the location of the file in my system but I am still getting the same error. Please help me out as how this error can be rectified.
[image]

YuxuanSnow, 2021-04-06:
Basically, you have to change the absolute path in .conf file. That means you have to change the "user" to your computer name.

Rishabh-Samra, 2021-04-11:
Yes. I had done that before but still getting the same error. Attaching the screenshot of my .conf file too where I had changed the path in line 15.
[image]

YuxuanSnow, 2021-04-11:
i looked your command window again and found that you didn't use the .conf file.  Actually you have all the arguments in the .conf file, you can just run roslaunch esim_ros esim.launch config:=cfg/unrealcv.conf to use the .conf file.  <br>
Do you want to run this example? https://github.com/uzh-rpg/rpg_esim/wiki/Photorealistic-Rendering-Engine-based-on-Unreal-Engine

Rishabh-Samra, 2021-04-11:
Thanks for your reply. So shall I follow the instructions of preparation
and then what you told from this link  (
https://github.com/uzh-rpg/rpg_esim/wiki/Photorealistic-Rendering-Engine-based-on-Unreal-Engine)
and then should simulate events with ESim as mentioned in this link:
https://github.com/uzh-rpg/rpg_esim/wiki/Simulating-events-from-a-video  ?

*Sincere Regards*
Rishabh Samra
*MS(Research) Scholar*

*IIT Madras*
*Home Page*:  https://sites.google.com/smail.iitm.ac.in/rishabhsamra/home

YuxuanSnow, 2021-04-11:
No, the link i have attached is an another tutorial. If you only want to simulate events from videos, you don't need to do this. Actually, you don't need any camera file if you just want to simulate events from videos. Have you downloaded video and do the preprocessing just as the tutorial?

YuxuanSnow, 2021-04-11:
Maybe it's caused by the wrong directory. Before you run "rosrun esim_ros esim_node", you have to change your directory by "roscd esim_ros". Then you are in directory "~/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros/"

Rishabh-Samra, 2021-04-11:
Yes I have downloaded that cheetah video and has done the preprocessing as
mentioned in that tutorial to get a sequence of image frames.
But then I am getting the error as posted in the issue when I am doing
rosrun by running this command(path of data_folder,out_bag modified ) as
follows: *rosrun esim_ros esim_node \ --data_source=2 \
--path_to_output_bag=/tmp/cheetah_example/out.bag \
--path_to_data_folder=/tmp/cheetah_example/frames0 \
--ros_publisher_frame_rate=60 \ --exposure_time_ms=10.0 \ --use_log_image=1
\ --log_eps=0.1 \ --contrast_threshold_pos=0.15
\--contrast_threshold_neg=0.15  *

What can be the problem with this?

*Sincere Regards*
Rishabh Samra
*MS(Research) Scholar*

*IIT Madras*
*Home Page*:  https://sites.google.com/smail.iitm.ac.in/rishabhsamra/home

YuxuanSnow, 2021-04-11:
Did you change the directory when you did "rosrun"?

Rishabh-Samra, 2021-04-12:
I think it should not matter which directory I did rosrun. But then also I tried to run in sub-directories of the path **~/sim_ws/src/rpg_esim/event_camera_simulator** but getting the same error as before of **Check failed: fileExists(FLAGS_calib_filename) Camera file does not exist** What can be done? 
@YuxuanSnow  Can I get your email id to ask a few related things through email?