uzh-rpg/rpg_esim #89: While simulating events from video, can only extract event data for 45 frames out of 960 frames from out.bag...

Rishabh-Samra, 2021-05-12:
I followed steps for simulating event data from video using running cheetah video(all steps mentioned in the link was followed). I got the output event data as out.bag file. Then I tried to follow instructions mentioned here to convert event data from bag file to yaml file:    [http://wiki.ros.org/rosbag/Tutorials/reading%20msgs%20from%20a%20bag%20file](url). 
I echoed(printed) topic: **/cam0/events/** (out of the topics mentioned in attached pic) to get events.yaml but unfortunately the yaml file has event data corresponding to the duration of first 45 frames only out of 960 frames.

I am not getting where exactly the data had gone missing? Any help to address the above issue is highly appreciated!!

[image]

shilpakancharla, 2022-03-10:
I figured it out. When using this step of the tutorial: `python scripts/generate_stamps_file.py -i /tmp/cheetah_example/frames -r 1200.0`, make sure you're setting the Hz value to whatever your fps is. In my case, it was 29.97 Hz, or fps, of my original video. So overall, just change that value.

Dengj618, 2023-12-14:
> I figured it out. When using this step of the tutorial: `python scripts/generate_stamps_file.py -i /tmp/cheetah_example/frames -r 1200.0`, make sure you're setting the Hz value to whatever your fps is. In my case, it was 29.97 Hz, or fps, of my original video. So overall, just change that value.

I'm having a problem with a timestamp that doesn't match. I did forget to set the frame rate here and found your answer very useful.