uzh-rpg/rpg_esim #90: different resolution has the same field of view

Jee-King, 2021-05-19:
Hi, thanks for your great work!

How can I get different resolution events but have the same field of view by setting image_height, image_width and data in .yaml?

Jee-King, 2021-05-20:
Hi, I change the values of the _data_ to have the same field of view.
For example, with data:[200,200,120,90], I can get the image with 180 height and 240 weight. When I want to obtain the image with 90 height and 120 weight, I change values of data to [100,100,60,45].  Finally, I can get 2643 images with the size 180*240, and 2512 images with the size 90*120.  When I change the values, it will make some errors as follows:

` root@f3e6dfcfbcde:~/sim_ws/src/rpg_esim/event_camera_simulator/esim_ros# roslaunch esim_ros esim.launch config:=cfg/example.conf
... logging to /root/.ros/log/37f14da4-b90c-11eb-b6e0-0242ac120007/roslaunch-f3e6dfcfbcde-6321.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://f3e6dfcfbcde:45135/

PARAMETERS
 * /optic_flow_viz/arrows_scale: 0.07
 * /optic_flow_viz/arrows_step: 7
 * /optic_flow_viz/arrows_upsample_factor: 1
 * /optic_flow_viz/publish_rate: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dvs_renderer (dvs_renderer/dvs_renderer)
    esim_node (esim_ros/esim_node)
    optic_flow_viz (esim_visualization/optic_flow_converter.py)

auto-starting new master
process[master]: started with pid [6331]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 37f14da4-b90c-11eb-b6e0-0242ac120007
process[rosout-1]: started with pid [6344]
started core service [/rosout]
process[esim_node-2]: started with pid [6356]
I0520 01:39:32.991345  6356 imu_factory.cpp:20] Initialize bias ...
I0520 01:39:33.014379  6356 imu_factory.cpp:27] done.
I0520 01:39:33.014412  6356 imu_factory.cpp:35] Initialize IMU ...
I0520 01:39:33.014420  6356 imu_factory.cpp:44] done.
I0520 01:39:33.015410  6356 data_provider_online_render.cpp:63] Horizontal FOV: 61.5044 deg
I0520 01:39:33.015463  6356 data_provider_online_render.cpp:64] Vertical FOV: 47.9772 deg
I0520 01:39:33.015475  6356 data_provider_online_render.cpp:65] Diagonal FOV: 73.223 deg
I0520 01:39:33.015489  6356 renderer_factory.cpp:75] Loading texture file from file: /root/sim_ws/src/rpg_esim/event_camera_simulator/imp/imp_planar_renderer/textures/office.jpg.
I0520 01:39:33.102098  6356 renderer_factory.cpp:85] Pre-filtering the texture with median filter of size: 13.
process[dvs_renderer-3]: started with pid [6362]
I0520 01:39:33.813468  6356 renderer_factory.cpp:92] Pre-filtering the texture with gaussian filter of size: 2.
process[optic_flow_viz-4]: started with pid [6363]
I0520 01:39:33.968417  6356 planar_renderer.cpp:34] K_src = 2269.92 0000000 0002080
0000000 2269.92 0001560
0000000 0000000 0000001
I0520 01:39:33.968570  6356 planar_renderer.cpp:36] T_W_P = 01 00 00 00
00 -1 00 00
00 00 -1 -1
00 00 00 01
I0520 01:39:33.973052  6356 ros_publisher.cpp:37] Initializing ROS
I0520 01:39:34.212764  6356 rosbag_writer.cpp:39] Will write to bag: /tmp/out.bag
W0520 01:39:34.212792  6356 adaptive_sampling_benchmark_publisher.cpp:57] Empty benchmark folder string: will not write benchmark files
W0520 01:39:34.213052  6356 synthetic_optic_flow_publisher.cpp:41] Empty output folder string: will not write synthetic optic flow files
I0520 01:39:34.213435  6356 rosbag_writer.cpp:275] t = 0.001 s
I0520 01:39:34.215504  6356 event_simulator.cpp:27] Converting the image to log image with eps = 0.001.
I0520 01:39:34.215670  6356 event_simulator.cpp:11] Initialized event camera simulator with sensor size: [120 x 90]
I0520 01:39:34.215685  6356 event_simulator.cpp:12] and contrast thresholds: C+ = 0.5 , C- = 0.5
W0520 01:39:34.215768  6356 camera_simulator.cpp:43] The images do not cover a time span long enough to simulate the exposure time accurately.
I0520 01:39:34.373281  6356 rosbag_writer.cpp:275] t = 0.501 s
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
publish image from callback
[ERROR] [1621474774.470953]: bad callback: <bound method FlowConverterNode._OpticFlowCallback of <__main__.FlowConverterNode instance at 0x7f80da02d7e8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/root/sim_ws/src/rpg_esim/event_camera_simulator/esim_visualization/src/py/optic_flow_converter.py", line 45, in _OpticFlowCallback
    if not self.first_msg_received:
AttributeError: FlowConverterNode instance has no attribute 'first_msg_received'

I0520 01:39:34.472385  6356 rosbag_writer.cpp:275] t = 1.001 s
[INFO] [1621474774.474708]: Started flow converter node
[ERROR] [1621474774.475263]: bad callback: <bound method FlowConverterNode._OpticFlowCallback of <__main__.FlowConverterNode instance at 0x7f80da02d7e8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/root/sim_ws/src/rpg_esim/event_camera_simulator/esim_visualization/src/py/optic_flow_converter.py", line 45, in _OpticFlowCallback
    if not self.first_msg_received:
AttributeError: FlowConverterNode instance has no attribute 'first_msg_received'

[INFO] [1621474774.475691]: Step size between arrows: 7
[INFO] [1621474774.476321]: Scale factor: 0.07
[ERROR] [1621474774.477322]: bad callback: <bound method FlowConverterNode._OpticFlowCallback of <__main__.FlowConverterNode instance at 0x7f80da02d7e8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/root/sim_ws/src/rpg_esim/event_camera_simulator/esim_visualization/src/py/optic_flow_converter.py", line 45, in _OpticFlowCallback
    if not self.first_msg_received:
AttributeError: FlowConverterNode instance has no attribute 'first_msg_received''

grandoba, 2021-09-23:
I got the same "FlowConverterNode._OpticFlowCallback" error as you have, but not because I changed anything. Anyways it is due to the python version. It should be python2 not python3. So change optica_flow_converter.py line 1 from `python` to `python2`.
```python
#!/usr/bin/env python ##wrong
#!/usr/bin/env python2 ##correct
```