uzh-rpg/rpg_esim #91: roscd: No such package/stack 'esim_ros'

aahuhu, 2021-06-09:
after i finished catkin_build esim_ros and add the path to /.bashrc ,this mistake occurred.

aahuhu, 2021-06-09:
i  commented out other workspace so that it made sense.

sefakurtipek, 2022-04-03:
How did you solve this problem?

mmsx, 2022-04-15:
Something wrong when you run catkin build~

ezragblock, 2024-04-18:
I wrote the command:
source ~/sim_ws/devel/setup.bash
and it worked

furkan-marker, 2024-05-09:
source ~/sim_ws/devel/setup.bash 
bu kodu hangi dizinde yazdınız?

furkan-marker, 2024-05-09:
ben yazıyorum ve hata çözülmedi

ezragblock, 2024-05-09:
I added it to the workspace main directory.

furkan-marker, 2024-05-10:
> I did it in the sim_ws directory.

Tesekkurler , siz demi esim ile uğraşıyorsunuz?

10 May 2024 Cum 07:44 tarihinde ezragblock ***@***.***> şunu
yazdı:

ezragblock, 2024-05-10:
I'm working with esim too, yes. I tried to do the color esim but it failed. It's full of bugs.
If you are working with the color feature and manage to get it to work, that'd be helpful.

furkan-marker, 2024-05-10:
I didnt understand your topic. My project is aruco/ qrcode detection with
esim. How do you process the event data?

10 May 2024 Cum 08:00 tarihinde ezragblock ***@***.***> şunu
yazdı:

furkan-marker, 2024-05-10:
The simulation is like this

şunu yazdı:

ezragblock, 2024-05-10:
Ah. I'm working on object tracking.

furkan-marker, 2024-05-10:
Do you know how i can process the event data ?

10 May 2024 Cum 08:47 tarihinde ezragblock ***@***.***> şunu
yazdı:

ezragblock, 2024-05-10:
There're many things to extract events from rosbag.