uzh-rpg/rpg_esim #105: How to use esim to simulate color events like CED dataset?

DachunKai, 2022-07-04:


eugenelyj, 2023-09-14:
Also working on this...Does anyone know?
I searched how to simulate the Bayer from a RGB image and found the code
```
import numpy as np # of course

im = cv.imread(...)
(height, width) = im.shape[:2]
(B,G,R) = cv.split(im)

bayer = np.empty((height, width), np.uint8)

bayer[0::2, 0::2]= G[0::2, 0::2] # top left
bayer[1::2, 0::2]= R[1::2, 0::2] # top right
bayer[0::2, 1::2]= B[0::2, 1::2] # bottom left
bayer[1::2, 1::2]= G[1::2, 1::2] # bottom right
```
So if my understanding is correct, then the event simulations are conducted on each of these Bayer channels. The results will be four event sequences but each only has 1/4 resolution.

eugenelyj, 2023-09-14:
@supitalp Could you help?

eugenelyj, 2023-09-19:
I find the solution is in the branch `feature/color` :)