uzh-rpg/rpg_esim #110: esim installation fatal error

2112jpc, 2022-09-28:
I am following the wiki instructions at github.com/uzh-rpg/rpg_esim/wiki/Installation.
I managed to install ROS without problems (I am using Ubuntu 20.04 and Noetic was recommended by the ROS site) and I installed catkin.
However I am getting the below error when I get to the “Clone this repository and run vcstools” step.  Any help would be appreciated.

cs@cs-l004:~/sim_ws/src$ git clone git@github.com:uzh-rpg/rpg_esim.git
Cloning into 'rpg_esim'...
The authenticity of host 'github.com (192.30.255.112)' can't be established.
ECDSA key fingerprint is SHA256:p2QAMXNIC1TJYWeIOttrVc98/R1BUFWu3/LiyKgUfQM.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,192.30.255.112' (ECDSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
 
Please make sure you have the correct access rights
and the repository exists.

JinghangLi, 2022-10-19:
Hi, maybe you can try to replace the link "git@github.com:uzh-rpg/rpg_esim.git"  with "https://github.com/uzh-rpg/rpg_esim.git" and git clone will work.

In addition to that, you need to change the URL in "rpg_esim/dependencies.yaml" file. I've posted the contents of my modified yaml file below, just replace the contents of the original yaml file directly. I hope this helps you.
```
repositories:
  catkin_simple:
    type: git
    url: https://github.com/catkin/catkin_simple.git
    version: master
  ze_oss:
    type: git
    url: https://github.com/uzh-rpg/ze_oss.git
    version: master
  gflags_catkin:
    type: git
    url: https://github.com/ethz-asl/gflags_catkin.git
    version: master
  glog_catkin:
    type: git
    url: https://github.com/ethz-asl/glog_catkin.git
    version: master
  eigen_catkin:
    type: git
    url: https://github.com/ethz-asl/eigen_catkin.git
    version: master
  eigen_checks:
    type: git
    url: https://github.com/ethz-asl/eigen_checks.git
    version: master
  minkindr:
    type: git
    url: https://github.com/ethz-asl/minkindr.git
    version: master
  minkindr_ros:
    type: git
    url: https://github.com/ethz-asl/minkindr_ros.git 
    version: master
  yaml_cpp_catkin:
    type: git
    url: https://github.com/ethz-asl/yaml_cpp_catkin.git
    version: master
  rpg_dvs_ros:
    type: git
    url: https://github.com/uzh-rpg/rpg_dvs_ros.git
    version: master
  assimp_catkin:
    type: git
    url: https://github.com/uzh-rpg/assimp_catkin.git
    version: master
```

2112jpc, 2022-10-20:
Thank you very much for the suggested fix.  I won't know when I will be able to try it out but I will keep it in my notes.  I will keep this issue open in case anyone else can comment on the fix.

Fulmen67, 2023-02-24:
I had a similar issue:

```
~/sim_ws$ git clone git@github.com:uzh-rpg/rpg_esim.git
Cloning into 'rpg_esim'...
Warning: Permanently added the ECDSA host key for IP address '140.82.121.4' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.
```

Resolved by following this [git tutorial](https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent). You need to create a local SSH key, add it to the SSH agent and finally copy it to your Github account to authenticate yourself.

Cheers,

Youssef